cleanflight/cleanflight #2091: naze32 in 1.12.1 firmware  with luxfloat pid can't arm

lit1088, 2016-04-29:
naze32 in 1.12.1 firmware  with luxfloat pid can't arm, but i change to multiwii(2.3) pid
 or multiwii(Rewrite) pid ,it can arm, what can i do with luxfloat pid for it to work well?

njozsef, 2016-04-30:
naze32 rev5 dof10. luxfloat is dont arm.

lit1088, 2016-05-03:
but sometimes i can arm, i don't konw what's matter whit it

lit1088, 2016-05-03:
but sometimes i can arm, i don't konw what's matter with it

WoooSaaaW, 2016-05-11:
I had the same issue with my naze rev6 board doing the same thing, it would load sometimes but the majority of the time it would not. Naze 6 has issues with luxfloat, you can get it to arm by doing the folloing " disabled the barometer with set baro_hardware = 1 and enable EMF avoidance with set emf_avoidance = on"

I discovered this fix here: https://github.com/borisbstyle/betaflight/issues/110 - unfortunately it still has issues even though it arms. As they mentioned in this forum it has errors and those errors cause glitches in the motor with damped light enabled which will cause your motors to heat up and i would not recommend to run them with damped light on due to risking your motor safety. Try Multiwii for now since luxfloat is buggy with naze6

thiagohmoreira, 2016-05-26:
I can confirm this also happens on my naze32 (afroflight32) rev5  with the 1.13.0-RC1 firmware. 

Did a clean firmware install (with full erase) and manually reconfigured the thing (without restoring my previous settings). With a specific channel or stick ARM it won't arm. 

Changed PID back to multiwii(Rewrite) and it worked just fine.

martinbudden, 2016-05-30:
This is not a bug, it's deliberate behaviour. Cleanflight now has a check to disable arming if the board is overloaded, see:

https://github.com/cleanflight/cleanflight/blob/master/src/main/mw.c#L263

Previously the board could arm, but this could then cause problems in flight.

I suggest you use PID_MWREWRITE, since is version 1.13 it is essentially the same as PID_LUXFLOAT, except that it uses fixed point arithmetic.

closed 2017-02-25 by hydra