cleanflight/cleanflight #2093: Naze32 acro rev.5 SBUS, MinimOSD,Bluetooth, GPS - hangs unarmed (three leds on)

8link, 2016-04-29:
New hexcopter on the way, configured and almost finished. 
During first tests I've noticed that after random (3 sec - 1 min etc.) amount of time, when FC is unarmed and almost for sure only when I am connected to MSP with Ez-Gui app through bluetooth, FC hangs. Three leds light up, game over, no connection. When Ez-Gui is connected and FC is armed i was unable to replicate the problem, looks like everything is stable. Disconnecting GPS and external Mag without changing any settings is not solving this issue. It is possible that when i am not connected with EzGui issue does not occur, but I need to double check this.  

--- Config ---
Afroflight Naze32 acro rev5
Cleanflight v1.12.1
External mag over I2C
UART 1 - GPS 57600
UART 2 - Serial RX - SBUS 
Soft UART 3 - MSP 19200 - minimOSD (configured for slow updates) 
Soft UART 4 - MSP 19200 - bluetooth 
RSSI over RC channel 

5v line checked and very stable 5,01v - 5,02v
 
[image]

SteveAmor, 2016-04-29:
Hi @8link sounds like a great support question for rcgroups. Please can you try there and only report bugs here. Please can you close this issue as it's not a bug for developers to fix. Thanks.

8link, 2016-04-29:
Thank you for your reply, but to be honest i suspect that it may be a bug, as everything is wired and configured correctly and behaves randomly and only when boards in not armed. Are you 100% sure that it is not a bug that has something to do with MSP over softserial?

8link, 2016-05-04:
I did check everything again, switched FC and I am now 99% sure that it is some kind of a bug, as the second FC behaves identically. To replicate the problem you need to have MW OSD 1.6 on Softserial 1 with 19200 bound  and Bluetooth on Softserial 2 with 19200 bound. Also you need to power up the FC and connect to MSP via bluetooth with EZ-GUI.

martinbudden, 2016-05-04:
It sounds like you are overloading the poor processor.

Could you try the same setup with a more powerful (F3 based) flight controller?

8link, 2016-05-04:
The "funny" thing is that it will not hang when FC is armed. You can use softserial with 19200 bound right? Loop time is 2000 and I can see that it keeps up. Event it is overloaded shouldn't it lose some cycles and not just hang?

hydra, 2016-05-07:
Please give more details about your GPS receiver.  Please provide a config dump.

We have quite a lot on at the moment.  Using two softserial ports was never reliable, especially for scenarios that use transmit AND receive on both ports.  Softserial implementation is OK for a low speed telemetry output (transmit only) and MSP 19200 bluetooth.

As @martinbudden suggests, using an F3 board is the way to go for the things you want to do.  You need more hardware UARTS.  Suggest SP Racing F3 EVO / MINI or similar as they also keep UART1 dedicated (shared on other boards).

hydra, 2016-05-07:
See also #177

closed 2017-02-25 by hydra