cleanflight/cleanflight #2094: #define VBAT_SCALE_MAX 255? can it be changed to 1100?

tetraDB, 2016-04-29:
Hi,

I think that this change (if it would work) would increase the options to use with the **vbat** pin (or FCs with an onboard voltage splitter). (hope i'm not on my own here =)

I'm using a SPRacingF3 FC on a 34volt (8S) system.
Apparently the vbat pin supports up to 6S directly.
I have a power distribution board designed for apm that supports up to 10S. The **pdb** has onboard current sensing and a scaled down voltage output.

The output voltage is scaled down by a factor of 10. This means that I can safely feed that into the vbat pin (max ~3.4v for 8S) with an appropriate **vbatscale** in the cleanflight configurator.

One snag, it maxes out at 255 and I need it to be ~1100 or there abouts.
Is it possible to have **VBAT_SCALE_MAX** value increased without causing problems elsewhere?

I can only see two alternatives to modifying **VBAT_SCALE_MAX**, one is I would have it to make an inline voltage splitter to scale the direct voltage, and not use the **pdb** voltage splitter.
Second option would be to modify the boards voltage splitter (that mite not even work) and would be super fiddly.

Unfortunately I wouldn't know where to start to compile and test my own modified version of cleanflight.

proggod, 2016-04-29:
You may want to look at my PR i submitted to raceflight related to this.   The code is hard coded to expect a 10:1 divider, i fixed this so something like you have should work, you would just need to add a voltage divider circuit to yours.

hydra, 2016-05-07:
yes, i think being able to configure the divider and an offset would likely be useful.  considering the amount of targets out there this would likely by useful and reduce target specific code.  Similar to how the current sensor configuration is done.

closed 2017-02-25 by hydra