cleanflight/cleanflight #2095: For discussion - scheduler, pid controller, gyro update and faster looptimes

martinbudden, 2016-04-29:
I'm putting up for discussion some thoughts I've had about the way the scheduler, pid controller and gyro update work.

These thoughts are still a "work in progress", but are sufficiently well formed to open to discussion, hence this "for discussion" issue. The ideas have developed from the work I have done on the scheduler, the pid contoller and also by the ideas expressed in @digitalentity  's iNav pid controller and @borisbstyle 's changes to the scheduler (in particular changes to `taskMainPidLoopChecker` and `taskMainPidLoop`).

## Background

Currently the primary task in Cleanflight is the `"GYRO/PID"` task which is invoked via the `taskMainPidLoopChecker` function. Assuming `gyroSync` is `on` this funtion:

1. busy waits for a gyro update
2. calls the `taskMainPidLoop` function

In turn the `taskMainPidLoop` (description below is much simplified)

1. reads the gyro
2. processes the other sensors
3. calls the pid controller
4. calls `mixTable` which calculates the motor outputs
5. updates the motors with the calculated outputs

There are two problems to this approach:

1. The `"GYRO/PID"` task is the dominant task, on a NAZE it can take 60% of the processor bandwidth (with 1000us looptimes)
2. The actions of reading the gyro, calling the pid controller and updating the motors are tightly coupled. This is problematic since the ideal frequencies for reading the gyro and updating the motors are different. It is also problematic because there can be the overhead of doing unnecessary PID calculations.

In betaflight, @borisbstyle somewhat decouples these actions, but the decoupling is partial and I think some other fundamental changes need to take place to allow a fuller decoupling.

## A proposed way forward

My thoughts on how to move forward came from a realisation I had when I was reading @digitalentity's iNav pid controller. Namely that the pid controller has state (which I guess is obvious) and that state is partitioned. In particular the state has three main components:

1. RC related state, ie `tpaFactor`, `horizonLevelStrength`, and what is currently called `angleRate` but would be better named `desiredRate`
2. Gyro related state, ie the value of the `ITerm` acuumulator (now called `lastITerm[]` and `ITermLimit[]`, previously called `errorGyroI[]` and `errorGyroILimit[]`), and the states of the `DTerm` filters
3. The actual calculated PID output, ie `axisPID[]`

The problem with the current implementaton is that the all three partitions of the states are recalculated whenever the pid controller is called: for very short looptimes this is a significant overhead. However:

1. we only need to recalculate RC related state when we get new input from the radio controller, this happens at a much lower rate than getting new gyro data
2. we only need to recalculate the gyro related state when we get new input from the gyro
3. we only need to make the final pid calculation when we are about to call the mixer (which happens more frequently than getting RC data, but less frequently than getting gyro data)

## The proposal

1. We more explictly organise the PID state data, much as @digitalentity has done for his iNav PID controller, but we do a bit more than this
2. We add two new PID functions something like
    1. `pidUpdateRcState` which is called whenever new RC data arrives, that is from within the `taskUpdateRxMain` function and
    2. `pidUpdateGyroState` which is called when we get new gyro data, ie in what is now called the `taskMainPidLoop` function
3. We add a new PID `calculatePID` function which is called just before we call the `mixTable` function and update the motors
4. We and a new task, `taskMotors`, which is run independently by the scheduler with a desired time that is detirmined the ESC/motor combination. This task essentially will call `calculatePID` followed by `mixTable` followed by `writeMotors`
5. With the above changes I think it will be possible to run the gyro at full speed even on F1 processors, since only a relatively small amount of processing occurs within the `pidUpdateGyroState` function. Additionally we could elect to keep a moving average of the gyro values for use in calculating the PTerm.

There are some details ommited, for example in flight calibration will change some PID state data, but that is a low frequency activity which can be easily accomodated. Also I have glossed over the annex code, but I don't think that will be problematic.

Note that this also fits in with @borisbstyle 's idea of dynamic P-values. The P-value is part of the PID controller state and could be updated at an appropriate point.

ledvinap, 2016-05-02:
@martinbudden:
One problem with current code is that multiple functions are mixed in PID function.
Common control approach is to separate different blocks:
- inner gyro PID loop (+ mixer + motor write)
- gyro setpoint generation
 - ANGLE mode - setpoint from estimated attitude (possibly using PID)
 - HORIZON - setpoint from RC + gyro + attitude (taulabs implementation decomposes multiwii 
controller into basic components)
 - setpoint for autopilot modes
- RC processing (expos, rates, .. ), generate (set of) variables that describe desired action (~ setpoint for controller)
- attitude estimation

The gyro PID loop should run at motor update frequency. Filter / interpolation should be used to convert gyro sample rate to motor update rate. This is trivial when there is integer number of gyro samples per motor period (thus the use of denominators), but it can be handled in rational cases too.

Also jitter (removed by busy loop in taskMainPidLoopChecker) is important only when outputting motor value.
Using short (< motor update frequency) delay to synchronize motor updates (possibly from interrupt, thus reducing mail loop timing requirements) will completely remove jitter.

borisbstyle, 2016-05-02:
martin i2c will still linit you goung faster than 4khz on naze32. You will
not be able to get it any better.

Total transaction time for gyro values only is already 130us

hydra, 2016-05-07:
I'm all for a clearer seperation of concerns.  At the moment doing this would be bad for any merges we might want.  I suggest we defer this until a later date, but by all means have a look into making some changes and see what blockers you encounter.

closed 2017-02-25 by hydra