cleanflight/cleanflight #2097: I2C implementation on f1 chips

gvuyk, 2016-04-30:
No problem, just a question
Is there a particular reason why the I2C implementation for stm32f1 chips is interrupt based even though the read/write functions wait on a loop for the operation to finish?

ledvinap, 2016-05-02:
IMO working code was used and simplified to fit Multiwii i2c interface.

It would be great to have interrupt-driven I2C .. But without RTOS it's not that easy to do right

jflyper, 2016-05-02:
I was also looking at the I2C code in conjunction with #2075 & #2062.

I was wondering if 125KHz interrupt is too much to handle, and started to dream about DMA'ing once the funky arbitration is over.

martinbudden, 2016-05-10:
@gvuyk , can you close this issue now?

closed 2016-05-10 by gvuyk