cleanflight/cleanflight #2103: Important PRs for the 1.13.0 release

martinbudden, 2016-05-05:
@hydra , you mentioned wanting to get v1.13.0 out as soon as possible. I'm all for that. However, there are a few important PRs that should not be omitted:

1. **Bug fix** - CC3D - PPM glitch fix, PR https://github.com/cleanflight/cleanflight/pull/1790
2. **Bug fix** - Changed `TASK_ACCEL` desiredPeriod to 1ms, https://github.com/cleanflight/cleanflight/pull/2076
3. **Documentation** - Update of PID documentation to reflect `PID_LUXFLOAT` changes, PR https://github.com/cleanflight/cleanflight/pull/2059
4. **Test code** - Updates for Junit xml unit test results and test coverage, PR https://github.com/cleanflight/cleanflight/pull/1693 (This has been hanging around for ages and @AndersHoglund has rebased it 4 times). We need to encourage test code related PRs.
5. **Cleanup** - Cleanup - further tidying of `config.c` PR https://github.com/cleanflight/cleanflight/pull/2077

1 and 2 are in betaflght.
We keep getting issues raised which are related to 2.
3 and 4 are entirely safe, since they are documentation and test code related.
5, while not essential, is a very good 'nice to have' and is further cleanup related to PR https://github.com/cleanflight/cleanflight/pull/2034 - Configuration storage - templates + improvements

mrzed72, 2016-05-05:
https://github.com/cleanflight/cleanflight/issues/1780
https://github.com/cleanflight/cleanflight/issues/2079

martinbudden, 2016-05-05:
@mrzed72 , both issues https://github.com/cleanflight/cleanflight/issues/1780 and https://github.com/cleanflight/cleanflight/issues/2079 should be fixed by PR https://github.com/cleanflight/cleanflight/pull/2076, item (2) in my list above.

hydra, 2016-05-05:
great, I'll crack on with this in the next day or so.  Really important to get the 1.13.0 and latest development configurator pushed out to the world, especially for USB VCP users.

hydra, 2016-05-06:
@martinbudden i merged 1 to 5. 6 needs a change.

The MSP cleanup PR should be merged for 1.13.0 IMHO. #2075

hydra, 2016-05-06:
Someone (@digitalentry) needs to fix warnings being generated by inclusion of new mavlink code from #806

martinbudden, 2016-05-06:
@hydra, the problem that @ledvinap mentions in https://github.com/cleanflight/cleanflight/pull/2077 is a pre-existing one (as he says), not a problem that the cleanup introduced. So the cleanup code should be merged. We should keep things atomic - cleanup should just do cleanup, not attempt to fix bugs at the same time (that is the path to the dark side).

digitalentity, 2016-05-06:
@hydra cleanup pending. I prefer to have things atomic - make it working and supported first. Optimise and cleanup later. I'm also stripping down the library to reduce the source code size.

martinbudden, 2016-05-06:
@digitalentity , a minor quibble, but the whitespace in your MAVLink code is not consistent - sometimes tabs are used and sometimes spaces. Can you convert it all to spaces, as per Cleanflight standard?

digitalentity, 2016-05-06:
@martinbudden That's MAVLink library code that has tabs and spaces mixed up. I'd like to avoid changing 3-rd party code.

martinbudden, 2016-05-06:
@digitalentity , I hadn't realised that. I agree, it's good to avoid changing 3rd party code.

BTW, is there a plan to merge iNav into Cleanflight for version 1.14.0 ?

digitalentity, 2016-05-06:
@martinbudden I'm going to start submitting PRs for INAV-specific features after 1.13 gets released.

martinbudden, 2016-05-06:
@digitalentity - brilliant. I look forward to seeing this.

digitalentity, 2016-05-06:
@martinbudden I doubt INAV will make it into CF for version 1.14 - too much to merge, but some of the changes (IMU, GPS driver) will likely be merged. :smile:

martinbudden, 2016-05-06:
@hydra , @ledvinap I didn't put the MSP cleanup PR https://github.com/cleanflight/cleanflight/pull/2075 on the list because I assumed it was a done deal. Yes, of course this should go in.

On reflection, it would also be a good idea to get AlienFlightF3 V2 support PR https://github.com/cleanflight/cleanflight/pull/1638 in as well - since this is a localised change and won't jeopardise the release.

martinbudden, 2016-05-07:
@hydra , a late entry: PR https://github.com/cleanflight/cleanflight/pull/2108 - Added PIDs to OLED display profile page. Added TASKS page.

This change is localised to the display code and so "safe".

Having the PIDs on the profile page is really useful (especially since they are now standardised between LUX and MWR). And the TASKS page is useful to check if changes to looprates etc have caused processor bandwidth problems.

hydra, 2016-05-07:
merged #2108 and 6 (#2077) now.

hydra, 2016-05-07:
@martinbudden I updated the task description.  I've rescheduled a few issues, will continue to review the list and see what I can get done.

martinbudden, 2016-05-07:
@hydra , great.

It would be good if you could merge in PR https://github.com/cleanflight/cleanflight/pull/2109 , this tidies the makefile a bit so that the targets are handled all in one place.

MJ666, 2016-05-07:
Have rebased #1638 again today.

hydra, 2016-05-07:
@MJ666 thanks. Will look now.

hydra, 2016-05-07:
added `* Code review and test #2112` to task.

@Pierre-A @ledvinap @martinbudden - can you help? If so comment on the issue when you start looking so that work is not duplicated.

hydra, 2016-05-07:
@MJ666 on second thoughts I will look when i get a chance, it's 00:19 here now and I'm tired!  Enough for tonight.

martinbudden, 2016-05-08:
@hydra , see PR https://github.com/cleanflight/cleanflight/pull/2113. This saves about 850 bytes of ROM on CC3D by limiting the max number of motors to 6 (there are only 6 PWM pins on CC3D boards anyway, so this shouldn't cause a problem for many people). This should provide enough to allow PR https://github.com/cleanflight/cleanflight/pull/2112 to build - Travis build says CC3D overflowed ROM by 210 bytes. [I haven't actually tested the 2112 build with my PR though.]

skaman82, 2016-05-08:
It would be also nice if some new LED functionality could make it in the new release. This has been requested long time ago and @gaelj has finally made this happen:

• Selectable Armed Colors (and other fixed colors)
https://github.com/cleanflight/cleanflight/pull/2053

• Support LED cmds GPS, RSSI & Blink
https://github.com/cleanflight/cleanflight/pull/2101
https://github.com/cleanflight/cleanflight-configurator/pull/379

The "Support LED cmds GPS, RSSI & Blink in configurator" PR is not quite ready yet, but I think @gaelj will finish this quickly.

hydra, 2016-05-08:
@skaman82 i thought i already merged selectable arming colors.  but yes they are often isolated so i can take a look.

hydra, 2016-05-09:
@skaman82 i remember now, i reviewed it but was waiting for some minor updates to the code.  #2053 was a continutation of #819

hydra, 2016-05-10:
@martinbudden PR's suggest to defer have been deferred.

hydra, 2016-05-10:
@martinbudden #2098 (blackbox enhancements) isn't ready.  there's discussion still.

hydra, 2016-05-10:
deferred #2109 to 1.14.0

hydra, 2016-05-10:
@martinbudden see https://github.com/cleanflight/cleanflight/pull/2117#issuecomment-218180529

hydra, 2016-05-10:
#2117 merged, flight tested on CJMCU and EVO.

hydra, 2016-05-10:
ok, only two outstanding items remain:

SPRacingF3MINI Parallel PWM support #2110
and new beeper configuration #2112 

I'm done for tonight, suggest if someone has time to review #2112 and test.  I will work on #2110 tomorrow and then tag master for v1.13.0-RC1

martinbudden, 2016-05-11:
Almost there! Great work @hydra

skaman82, 2016-05-11:
@hydra https://github.com/cleanflight/cleanflight/pull/2101 is also good to go now

hydra, 2016-05-11:
ok, all done.  couple of warnings in master to fix.

hydra, 2016-05-11:
done, see 1e14fd70d6cb497cf7a1cf02a853eb4b1d716227

digitalentity, 2016-05-11:
@hydra you can merge #2107 for now - it reduces number of source files and has a temporary fix for mavlink warnings.

hydra, 2016-05-12:
@digitalentity for RC2, yes.  @martinbudden can you update the topic description to keep track of bugs raised against RC1 so we all know what to do for RC2 ?

No new features accepted until v1.13.0 is FINAL.  Bug fixes to master only.

digitalentity, 2016-05-12:
@hydra great! I have a bunch of things for next 1.14 relese.

martinbudden, 2016-05-12:
@hydra , I couldn't think of a better name for topic, since it's still about preparing for the v1.13.0 release. I've added a section **Release Candidate bug reports** though.

hydra, 2016-05-12:
Release notes updated.

https://github.com/cleanflight/cleanflight/releases/tag/v1.13.0-RC1

hydra, 2016-05-12:
Links posted on social media:

https://twitter.com/Cleanflight/status/730697489582592001

https://www.facebook.com/permalink.php?story_fbid=855083651268087&id=582474325195689

Please share!

hydra, 2016-05-12:
Can someone create an issue in the configurator repo and collate PR's they feel should be merged?

hydra, 2016-05-12:
I've just pushed out a small release of the configurator v1.2.2 with the current development head that I've been using for quite a while now.  Mainly so that VCP users get the benefit.  Now we can concentrate on configurator v1.3.0.

hydra, 2016-05-29:
want to merge in #2139 but there is an issue with the blink code.

see https://github.com/cleanflight/cleanflight/pull/2139#issuecomment-222388359

hydra, 2016-05-29:
@martinbudden thanks for collating the RC bug report issues.  much appreciated.  you're doing awesome work.

martinbudden, 2016-05-30:
@hydra , glad my work is appreciated. It would make things easier for me if you gave me permissions to change the labels on issues and PRs, and also to close issues (there are lots of "support questions" issues that can be closed). There are also quite a few forgotten PRs that could do with relabelling.

hydra, 2016-05-30:
@martinbudden sure, will do (if i can work out how..)

hydra, 2016-05-30:
RC2 is out now.

Thanks everyone!

The plan now is to wait for RC2 bug reports and release final code on 6/6/2016.

martinbudden, 2016-06-01:
@hydra , thanks for adding me as a Cleanflight member. I've made an initial pass through the PRs and some of them labels. Two questions:

1. There are quite a few documentation PRs outstanding. Is it OK for me to merge these in?
2. There are quite a few minor enhancements, including a few old ones that seem to have been lost in the large number of PRs. Is it OK for me to assign these to the 1.14 milestone? (Of course on further examination they may be bumped to a later milestone, or dropped entirely)?

hydra, 2016-06-05:
1. with regards to documentation PRs ensure that all links are relative before merging (or fix locally via another comit locally before merging).  Inform original authors via commit links any changes made so they can contribute better in the future.
2. With regards to other minor enhancements and milestone.  Assign minor enhancements to 1.15.0, not 1.14.0.

I have opened #2211 to discuss 1.14.0 and the future.

hydra, 2016-06-10:
Final is out (Since 6/6/2016).  It was late, i didn't have time to update this and have been working on the OSD code all week.

closed 2016-06-10 by hydra