cleanflight/cleanflight #2104: CJMCU gyro and mag problem

Benik3, 2016-05-05:
Hello.
I found some problem with gyroscope and magnetometer on mini board CJMCU.
Accelerometer is OK from default, but gyro axis are totally wrong (from the graphs):
Pitch nose up - Y to negative
Roll right - X to positive
yaw to right - Z to negative
I wasn't able to fix it by settings in Configuration. Always one axis remain in opposite direction.

Magnetometer doesn't show any changes in axis X and Y. In axis Z there are changes.
But it's interesting, that on animation on setup page recognize where is head and it changes when I YAW the quad in my hand.

There is no specific HW version on the board, but it should be the v2 (with jumper for boot0). I'm on latest version 1.12.1

Thanks for any help

Benik3, 2016-05-06:
Just tried to flash old 1.5.0 firmware. Same problem.
I'm starting to think if it's not HW problem or GUI...

hydra, 2016-05-07:
i have two CJMCU's here, rev1 and rev2 and they both work with older versions of cleanflight.  I have not had chance to test them with the latest master branch/development builds.

If even the old version doesn't work it's likely it's not a software issue.  Can you please attach photos of your CJMCU board to this issue?

Did you try using `mag_align`, `gyro_align` and `acc_align` cli commands?  A video might help too.

Benik3, 2016-05-08:
Thanks for reply.
Well I was also thinking about HW problem.
But is interesting, that acc is working fine (and gyro is on one chip MPU6050).
Also on Setup page mag looks working. How should it act in sensor tab on axes X,Y,Z?

I tried to build latest master version of CF and also download latest GUI. Still the same.

As I wrote, I tried to fix it with Gyro_align, but always one axis remain opposite (the closest are CW270 (opposite Z) and CW270 flip (opposite Y)).

I will try to do more research
Here are photos of the board
[image]
[image]

hydra, 2016-05-11:
yeah, that's a very early version it seems.  i've not seen one with component markings on before.

i test flew latest master on rev 1 and rev 2 boards yesterday, no problems.

not really got time to look into this at the moment.

please continue to research the problem.  please reference the datasheets for the mag and you should be able to verify the alignment of the sensor and pick the appropriate alignment.  don't forget to recalibrate mag after changing mag alignment too.

keep me posted.

Benik3, 2016-05-12:
Hello.
Thanks for reply. Can you please send me some detailed photo of the boards you have, so I can see markings on components and tracing?
I will try to do some research...
I can't say if it's newer or older release. I bought it on aliexpress. It's little hard to buy some of these board with good price...
Thank you

MJ666, 2016-05-13:
Orientation of the sensors is the same as on my CJMCU boards. Also the layout does not really look different except the additional pads for the boot jumper. As long the firmware is working on this board is schuld work. Since Gyro and ACC are in the same chip it even would not make any sense the orientation should be wrong (or different).

Benik3, 2016-05-13:
Thanks for reply.
Yeah that's exactly what I thought.
When you connect the board to configurator, what version do you have in log console?
I tried yesterday to force FW to Rev_2 by editing hardware_revision.c file, but it still said rev 0... :/

Arakon, 2016-05-14:
I have the same board here. It's showing the same behavior, although it seems to be flying just fine.

Benik3, 2016-07-12:
So finally I was able to test more the board.
It looks flying well in level mode, so the acc is probably wrong only in GUI.
In version 1.13.0 it's detected as Rev1 board.

closed 2017-02-25 by hydra