cleanflight/cleanflight #2105: Value of `acc_1G` hardcoded to 256 in `sensors/acceleration.c`

martinbudden, 2016-05-05:
In `sensors/acceleration.c` `acc_1G` is hardcoded to 256, instead of using the value
```
extern uint16_t acc_1G;
```
That is set by the device drivers.

However, I'm not sure if this is deliberate or a mistake.

ledvinap, 2016-05-05:
`sensors/acceleration.c` owns the value and sets default. All other modules use `extern`, drivers update the default.

ledvinap, 2016-05-05:
BTW: Instead of `acc_1G` it would be better to scale ACC value to common precision (in accelerometer drives). `accADC` is 32bit anyway (to avoid overflow after applying offset ; maybe a bit of overkill), so we may convert the code to assume `acc_1G = 16384` (maximum sensitivity of mpu6050, far higher than neccesary) or `acc_1G = 4096` (largest in current code)

martinbudden, 2016-05-05:
@ledvinap , I think you are right, scaling the ACC value makes sense.

I think it is good to use 32bit values for `accADC`, it's a very small amount of memory and if overflows occur they are a nightmare to track down. I think we should also go 32bit for the gyroADC raw values - Raceflight already has an overflow problem here.

Certainly having a the device drivers use an extern that is declared in high level code is horrible.

I was attempting to clean this up, see https://github.com/cleanflight/cleanflight/compare/master...martinbudden:tidy_acc_1g?expand=1

Do you have any views on that?

ledvinap, 2016-05-06:
Gyro is `int32` too, both were changed to fix overflow problem.

One advantage of 16 bit on Cortex-M3 is considerably faster multiply/MAC instructions (http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0337h/CHDDIGAC.html). This may be significant if we want to implement longer FIR filters.

Current gyro noise is tens of count peak-peak (cleanest intervals), dividing gyro data to -(2^14) - 2^14 range (dropping one bit from 16bit MPU6050 range) will work fine.

Acc noise (from borisbstyle log) is well over 1000 peak-peak, so we can drop bits easily.

ledvinap, 2016-05-06:
@martinbudden : cleanuing up `acc_1G` if IMO not worth the effort. It's much easier to remove it (or keep it as private scaling factor inside driver)

martinbudden, 2016-05-06:
@ledvinap , agreed. Closing.

closed 2016-05-06 by martinbudden