cleanflight/cleanflight #2106: Change BMP280 to MS5611

enrique15241524, 2016-05-06:
Hi my friends;
i have Naze32 acro rev6
board have BMP280 barometer
but i did not like to stable
i want external MS5611 barometer
how change default to ms5611 on CLI?

sory my bad english :)

regards
Murat

skaman82, 2016-05-06:
Hi, this is not a issue - it is a support question. Please ask this type of questions in the RCGroups forum and close this issue. You should also look up the CLI command baro_hardware.

swilkens, 2016-05-06:
Seems that option isn't in the docs for the CLI commands!

Anyway, the MS5611 can be set with option 3 for baro_hardware:

`set baro_hardware = 3`

See here:
https://github.com/cleanflight/cleanflight/blob/e133331cf34f94109774181c2d6f3ae74f4fba05/src/main/sensors/barometer.h#L24

enrique15241524, 2016-05-06:
Thank you swilkens...

closed 2016-05-06 by enrique15241524