cleanflight/cleanflight #2118: SmartPort for F1 targets on real UART

elepete, 2016-05-08:
Hello

as SmartPort only works with SoftSerial on F1 targets. An Inverter on a single wire to single wire UART does not work - at least I could not come to an idea (this would result in positive feedback - an flipflop circuit).

Proposal:
in smartport.c the portOption = Serial_BIDIR is set (makes sense for F3 targets)
change in file serial_uart_stm32f10x.c:
--- remove ---
if (options & SERIAL_BIDIR) {
  gpio.mode = Mode_AF_OD;
  gpioInit(USART3_GPIO, &gpio);
---
This makes the UART having RX & TX on separate pins so that the inverter circuit with 2x BJT or FET is useable.

Could this be tested / implemented?

Best regards - Peter

hydra, 2016-05-12:
what is stopping you testing this yourself and submitting a pull request?  seems like the perfect small task to get started on.  see the docs/developer for help on getting started.

elepete, 2016-05-14:
thank you - I needed a kick
I expected setting up the toolchain under Debian would be difficult - however it was not that compicated (BUT - need to use "gcc-arm-none-eabi-4_9-2015q3", otherways VCP does not work - reading and believing helps ;-)

So, to have SmartPort on a CC3D I have changed file "smartport.c".
Following little tweak is needed:
---
#ifndef CC3D
    portOptions = SERIAL_BIDIR;
#else
    portOptions = 0;
#endif
--- 

The #ifndef list could / should be continued for more F1 targets to get Smart.Port working on other boards with inverter as well.

What needs to be done to get this in the official sources?

Peter

ledvinap, 2016-05-14:
@elepete : with #2120 you can use gcc 5.3 ..

martinbudden, 2016-05-15:
@elepete , to get this into Cleanflight you need to make a pull request. It's too late for v1.13, but it could go in v1.14

closed 2017-02-25 by hydra