cleanflight/cleanflight #2132: FlySky channel input problem naze32

judletfpv, 2016-05-13:
I have my Aux1 and Aux2 switches working fine. the problem is my sticks don't give any input to the receiver on naze32 (rev5) I am using PWM and 6 channels. My radio is the fs-i6 and my receiver is the fs-iA6.

my throttle is stuck at: 885

and all the other inputs are stuck at 15000.(not AUX1 or AUX2)

thanks to everyone that can help.

Arakon, 2016-05-13:
This is a support question, not a software bug. Your channels are in the wrong order or out of range if this happens, it's going into failsafe. Set up your endpoints on the transmitter.

closed 2016-05-13 by hydra

judletfpv, 2016-05-14:
Yes, my endpoints are set up. but no input/ signal is coming from the sticks. it will still only arm, and the motors will spin when armed.

marcoberri, 2016-05-15:
I have same problem with spracingf3. Help please!