cleanflight/cleanflight #2136: Naze32: moving LED_STRIP away from SOFTSERIAL

KP1533TM2, 2016-05-14:
Good news everyone!
My PRM-02 arrived today, and I've decided that (for now, atleast) I don't want to do a direct connection I initially wanted ([#1690](https://github.com/cleanflight/cleanflight/issues/1690)). MavLink telemetry works great, thanks to whoever implemented it :+1: 
To cut a long story short: I want to setup my Naze32 with GPS and PRM-02 on softserial ports, my Radiolink SBUS receiver on USART2 and my Bluetooth on USART1. While there's no problem with the latter two, I've ran into an ages old bottleneck where I can't enable SOFTSERIAL along with LED_STRIP (that LED strip that I don't want to get rid of - I didn't even get to use it yet).
So I moved LED_STRIP out of the way and placed it on PWM2. Now, instead of conflicting with SOFTSERIAL, it conflicts with RSSI_ADC. But that doesn't bother me much, since in my setup that feature is useless anyway. That way, I have my LEDs working and I can see all telemetry stuff on my Radiolink AT9 TX :smiley:.
So, for the next few days I'm going to tidy up and test my changes locally (my GPS is extremely flaky and I don't know why; granted, my quad is not finished yet and I'm testing it by just placing it near the window, so maybe it's nothing), and so the question I want to ask - would anyone else be interested in such a change for the Naze32?
Maybe even someone could help me test the changes in action, since as I said, I don't have a 100% completed quad myself yet.

skaman82, 2016-05-14:
Great +1 for Softserial AND LED Support.

gaelj, 2016-05-15:
+1 if you make it a configurable feature.

KP1533TM2, 2016-05-15:
I've been looking into the sources and figuring out how I can make a feature configurable, and came up with idea of having a variable (CLI-configurable, atleast for now) that would decide on LED channel placement. So, the setup would go somewhat like this:
```
# feature LED_STRIP
# set led_remap = 1 // For example, 0 for initial placement (instead of SOFTSERIAL), 1 for input 2 (instead of RSSI_ADC), etc...
```
...or something like that. That way, it would be still compatible with current configurator (that doesn't have that variable in GUI anywhere yet) and allow for things to be configured through CLI.

P.S. I've looked a bit into how features are handled in code now, and that system looks a bit inflexible and broken to me. Things are somewhat simple, but unnecesarily hardcoded... I have a few interesting ideas of my own how I would like to see feature allocation, but I'll leave them for another issue/request.

gaelj, 2016-05-15:
You could use strings instead of int in the CLI values, to improve clarity (the drawback is that currently users need to read the documentation to know what values are available, but this is also being improved)

KP1533TM2, 2016-05-16:
I finally made some progress making LED_STRIP configurable.
It's now in my fork [here](https://github.com/KP1533TM2/cleanflight/tree/led-strip-and-softserial), if you want to have a play around.
It is configured in the same exact way I described a day before. A table of strings wouldn't be hard to implement, it's just not the top priority. I tried to unify things a bit and created a table of 'LED strip interfaces', where each (or single) entry describes GPIO, DMA and TIMer used by LED_STRIP feature. That might help simplify LED_STRIP hardware initialization, but I haven't bothered with that yet.
The bad news is that the entire thing (as of current feature system) is kludgy and ugly (imho), especially in configuration validation 'department'. So much that idea of having two NAZE targets with different 'skillset' doesn't look that crazy to me anymore :smiley:. `config.h` is like hardcoded `#define` hell to me.
Also, I had to disable BLACKBOX to make firmware fit, so there's that...

Offtopic: can someone check how GPS works at 57600 baud on softserial along with something else on other softserial? Today my GPS receiver somehow forgot its 57600 baud setting and fell back to 9600 baud. 3D fix became rock solid, while it was getting annoyngly lost and found all the time at 57600. Hmm. :confused:

gaelj, 2016-05-16:
There's also some interesting stuff going on in PR https://github.com/cleanflight/cleanflight/pull/2139.

aequitas, 2016-05-23:
@KP1533TM2 trying out your branch on a Naze32 rev5. Looking good so far.Hoping to get GPS+SPORT working together with this.

KP1533TM2, 2016-05-24:
>Update: The SPORT not working is due to the GPS softserial being on a different baud. 

That limitation is described in the documentation, but it got me too at first :smile: 
I've tried to setup my GPS receiver to use 57600 (so it would work on the same speed as MavLink telemetry), but it seems that it doesn't save its configuration and defaults to 9600 all the time. That is, it works at 57600 baud until power cycled. It's an u-blox 7 without any permanent memory to store settings, or so it seems. 

Infact, that's another feature that would probably be useful, but would be too hacky to implement (for now, atleast): GPS initialization on power-up. It would be nice if Cleanflight could send some command to GPS (i.e. port config) at some initial speed (i.e. 9600) and then setup that serial interface according to configuration.

aequitas, 2016-05-24:
Forgot to test yesterday with stock cleanflight, but thanks for confirming my suspicion :).

I had the same idea yesterday and did some investigation. Turn out there is already some code that does this: https://github.com/cleanflight/cleanflight/blob/master/src/main/io/gps.c#L119-L126

Don't think it will work with every GPS, maybe most UBLOX. At least not mine, it's an old Nemerix NMEA bluetooth module. It used different registers for its settings. I will see if it makes sense to create a patch for that.

KP1533TM2, 2016-05-24:
Hm, interesting. Now I think that the way how softserial initializes along with GPS should be investigated too. Given the 'same baud rate on both ports' limitation of softserial, I now wonder what happens when both ports are enabled, but one of them starts changing baudrate (is that what happens upon GPS initialization?) without the other.

aequitas, 2016-05-24:
Had to hack RX into my GPS but managed to get Baud init working, was pretty simple: https://github.com/cleanflight/cleanflight/compare/master...aequitas:gps?expand=1

Both GPS and Telemetry working. I will try convert this into a configurable option when I find the time.

KP1533TM2, 2016-06-05:
Well, I've just assembled my quad yesterday (without GPS yet, though) and with all the TELEMETRY, SOFTSERIAL and LEDSTRIP enabled, and I was unable to arm. It seems that system load is too high, since when I turn TELEMETRY off, the load drops below 1 and it starts to arm.
With SOFTSERIAL and LEDSTRIP on, load is around 0.4-0.3.
With added TELEMETRY on top of that, load is around 1.5-2.
(PRM-02 module sits on SOFTSERIAL port)

I've set gyro_sync_denom = 2 as mentioned in [the other issue](https://github.com/cleanflight/cleanflight/issues/2192#issuecomment-223071503).
Load went to 0.2-0.3 with all the features on, and I was able to arm again.

I crashed once, relaxed my PIDs and tried to float it in my living room, but it's kinda jerky on descend. I'm not sure if that's related to anything, but there's that.

Floyer007, 2016-07-23:
That's doesn't sound good. 
Did you worked on in again? 

Another Question: Does the LEDstrip also work with the "spektrum sattelite" pin? 
Because that's a pin I don't need.

gaelj, 2016-07-23:
@KP1533TM2 Jerkiness on descent is often related to I being too high (AFAIK)
@Floyer007 LED strip only works on the dedicated pin.

KP1533TM2, 2016-07-24:
I haven't worked on it since, I moved to another frame and I don't have LEDs there yet. Just learning to fly now :)
@gaelj Thank you for the suggestion, I'll check it out.

McGiverGim, 2017-02-04:
Very interested in this. I suppose nobody is making progress?

closed 2017-02-25 by hydra