cleanflight/cleanflight #2140: Dangerous Default "Minimum throttle" value

Petfrogg, 2016-05-15:
The "Minimum throttle" should by default be set lower. I suggest we use the same numbers as we got as default minimum command.

The reason for this is that since MOTOR_STOP by default is not enabled and if your motors kicks in around 1040 the throttle when armed by the default would start at 1150. This is dangerous if you accidentally arms the quad with props while configuring it.

rb1205, 2016-05-30:
It depends on the ESC I guess, mine start rotating at 1120. IMHO it's less dangerous to have (relatively) high idle rpm than risking to have stopped motors while armed at default settings. Besides, if 1150 throttle is enough to be dangerous, then you're dealing with a very powerful setup where extra caution is required anyway.

hydra, 2016-06-05:
would like additional feedback from others on this.

most esc's i try kick in at around 1080-1090.  my initial thought is the same as @rb1205

MJ666, 2016-06-05:
I have Kiss ESC on nearly all my builds. There was never a problem with the actual defaults and all motor running smothly already with this value. Its also not an issue on brushed copters. I understand this is often an issue with various other ESC motor combination. Also the practise to not calibrate ESC an set the values manualy may play into this. Most ESC's don't have an crystal and the accuracy and sometimes the resolution of the detected PWM signal is not the same be ween ESC's. Incresing this default will help to make the problems less hurting.

Floyer007, 2016-06-06:
I'm using the AFRO-ESC with an minimum throttle of 1080.

Shoudn't everybody change the default value while their first setup?

closed 2017-02-25 by hydra