cleanflight/cleanflight #2143: cli command small_angle

taileron, 2016-05-16:
small angle is fixed to 25 degrees even with settings above in CF 1.13 RC1.
Accidentally disarmed vehicles can not be armed inflight again ...

andre92, 2016-05-25:
there is a new variable named max_arm_angle, default at 25, try that one

taileron, 2016-05-26:
thanks ... great now it works again.
Is there a trick to arm with higher throttle values as well?
If an esc cuts down by over current or another reason, to recover it has to be disarmed, how can I arm again as fast as possible to prevent the crash ?

andre92, 2016-05-27:
I'm afraid that is not possible. Cleanflight refuses to arm with throttle above min_check. If you want to rearm the craft in the air, you need to bring the throttle all the way down, then disarm and arm again. Disabling feature MOTOR_STOP should help if you try this (so motors always spin at idle when armed instead of stopping when you lower throttle below min_check)

taileron, 2016-05-27:
o.k. the switch in my new tx can do 1/10 of a second minimum throttle before arming.
1.13 RC1 with Lux Float flies absolutely great and is easier to tune than before.
With the higher P number there are less coarse P value changes. 
And max_angle_inclination has a much wider usable limit area.

(comparing to 1.12)

closed 2016-05-27 by taileron