cleanflight/cleanflight #2144: Newb, cant figure out how to save settings, CC3D?

dacobi, 2016-05-16:
Hi

Just flashed my cc3d with clean flight 1.10.0 and figured out how to connect using the STM virtual com driver, but how do I save settings?
When I press save and reboot I have to remove power and USB and reconnect again for the configurator to reconnect and when that's done non of my changes remain?

My primary focus is to be able to connect a DSM satellite to the main UART. Can this be done?

Kind Regards
Jacob

gaelj, 2016-05-16:
Try to flash CF 1.13-RC1, then use RC groups forums for help on configuration.

SteveAmor, 2016-05-17:
Hi @dacobi please can you use the support forums like RCGroups and close this issue as you are not reporting a bug.

closed 2016-05-17 by dacobi