cleanflight/cleanflight #2149: Headfree Mode - Tracks Orientation Incorrectly

kgustine, 2016-05-17:
Running V. 1.13 RC1 on SPRACINGF3

When I switch to headfree mode, pitch and roll act a little strange while still facing my original heading.

After I rotate 90 degrees either way and give it forward pitch, the quad throttles up! I then have no control of pitch what so ever.  Found this at work and didn't have time to create a blackbox log, but will later if no one else has a chance.

I was flying yesterday with 1.12 on this quad and the issue did not exist.

andre92, 2016-05-26:
I just wrote a patch for this: https://github.com/cleanflight/cleanflight/pull/2171 the code was changing commands for roll and pitch every loop (based on the ones from previous loop and not fresh from receiver)

hydra, 2016-05-29:
fix merged in #2176

closed 2016-05-29 by hydra