cleanflight/cleanflight #2153: Naze32 with Spektrum SRXL implementation

DaleKramer, 2016-05-18:
I can not get the SRXL port on my Spektrum AR9020 receiver to be recognized by my Naze32 rev6.
I have tried using flashing Naze 1.12.1 
I can attach a Spektrum satellite to the Naze32 port for it and the system works but I do not want to lose the redundancy of the 3 antennas on my AR9020.
I have sent the SRXL data line to pin 1 and pin 4 of the Naze32 and have tried XBUS_Mode_B provided (actually tried all providers :) with no luck.
The Sprektrum SRXL line seems to be 18 bits with the last 2 bits a CRC checksum that nobody was been able to figure out yet.  The first 16 bits seem to simply be the standard 16 bit Spektrum satellite data stream which Naze can already parse.
Could someone add a 'Serial Receiver Provider' to Cleanflight and add a NAZE firmware version that would parse the Spektrum SRXL data line?
It looks like you could use code already in the firmware for the Satellite but just ignore the last 2 CRC bits as nobody seems to be able to use them anyway.
I have found this code that would seem to parse the Spektrum SRXL line, if it would help:  
[http://www.rcgroups.com/forums/showpost.php?p=34160865&postcount=77](url)  
Thanks,
Dale Kramer

DaleKramer, 2016-05-19:
Sorry folks, this appears to be a NON-ISSUE.

I am not totally sure of what caused my sudden success, but Cleanflight now sees my AR9020 Receiver with its SRXL line going to pin 4 of the Naze32 rev6 (UART2).

I think it had to do with the fact that I flashed NAZE 1.13.0-RC1

SteveAmor, 2016-05-19:
Hi @DaleKramer glad it resolved please can you close this issue

closed 2016-05-19 by DaleKramer