cleanflight/cleanflight #2155: Tidying up JR DMSS satellite protocols

andypnz, 2016-05-19:
There are some issues with JR DMSS satellite implementation. Works with RA01 series sats with white PCB. Does not work with older RA01 series sats with black/brown PCB, or any of the newer RA02/03 sats.

JR are willing to share their protocols and help with testing - but are looking for someone to tidy this up.

Any ideas?

TomTuckers, 2016-05-21:
I would love this to be solved was looking forward to using the new RA02TL on my quad using cleanflight but its looking like its not possible 
??

ChrisZydel, 2016-08-26:
It would be fantastic to get this implemented, the RA02TL is much more compact than the RG612BX not to mention half the price. Do you have the protocol specs from JR? It would be great if we can compare the RA01TL and RA02TL to see what is different.

closed 2017-02-25 by hydra