cleanflight/cleanflight #2156: SPRACINGF3MINI air mode not lighting up in Mode in v1.13.0-RC1

divenet, 2016-05-19:
Is it my wrong doing somewhere in my settings or is there an issue with SPRACINGF3MINI air mode not lighting up in Mode in v1.13.0-RC1?

My modes are all set to aux 2, they light up green when activated in all other modes but not air mode during my test.

Thanks for any feedback

divenet, 2016-05-20:
@JamesHarrison,  shoud i close this issue if you have this in hand?

martinbudden, 2016-05-21:
@divenet , don't close this yet - issues are closed when the fix gets merged in.

divenet, 2016-05-21:
@martinbudden thanks, just let me know when its a good time to do so as i am a new user of github and still learning my way around and trying hard to become a valuable contributor.

Trying my best to help.  JL

hydra, 2016-05-30:
fix in #2159 merged.

closed 2016-05-30 by hydra