cleanflight/cleanflight #2161: com port not working

dhackens, 2016-05-22:
error message ... failed to open serial port

SteveAmor, 2016-05-22:
Hi @dhackens please can you use forums such as RCGroups for such support questions as this is not a bug for developers to fix. Please can you close this issue.  Many thanks.

S-T-A, 2016-05-25:
OS X El Capitan Version 10.11.4
where do get a drive to get the com to work???

This was the only solution:
CP210x Macintosh OS X VCP Driver v4 Release Notes
Copyright (C) 2015 Silicon Laboratories, Inc.

This release contains the following components:

* SiLabsUSBDriverDisk.dmg - Image containing the VCP Driver Installer
* ReleaseNotes.txt (this file)
* uninstaller.sh - a bash shell script for removing the driver

Known Issues and Limitations
----------------------------

This release includes the Macintosh OSX driver for the Intel and
PowerPC Platforms versions 10.5, 10.6, 10.7, 10.8, 
10.9, and 10.10.........
--------------------------------

What am i so pose to do?

hydra, 2016-05-27:
closing, see comment from @SteveAmor

closed 2016-05-27 by hydra