cleanflight/cleanflight #2162: motor layout wraong after board orientation change (naze32)

labomat, 2016-05-23:
I changed the board alignment of a naze32 acro v5 to Yaw 90° (board 90° CW) and after that motor 1 & 2 are interchanged and both turn the wrong direction... After changing the board alignment to normal everything is okay again (cleanflight/naze 1.12.0). 
I also have enabled the gimbal servo option - using servo output 1 for the servo and 3 to 6 for the escs (so output 3 and 4 are wrong).﻿

hydra, 2016-05-23:
bug already fixed in 1.12.1.  please report bugs using the latest version.  please test the latest 1.13.0-RC1 release candidate and report back please.

labomat, 2016-05-23:
I upgraded to 1.12.1 and it doesn't change. I made a short video to show the problem: https://www.youtube.com/watch?v=RjTnjPWnPwA
You can see ( a bit blurry):
It's version 1.12.1.
Board orientation is 90°.
When operation motor 1 motor front right is turning - when operation motor 2 motor back right is turning.
Pushing roll stick to left - left indicator led is blinking.
Pushing roll stick to right - both indicator leds are blinking

Before making the orientation change I already was flying the quad successfully.

hydra, 2016-05-27:
please provide config dump.

hydra, 2016-05-27:
if anyone else has time to confirm and investigate this it would much appreciated.

hydra, 2016-05-30:
motor order does not change when board orientation is changed.  board orientation only effects the output of the sensors. 

there was an issue with the motor mapping on the naze but that is unreated to board orientation.

hydra, 2016-05-30:
@labomat basically, i think you have just wired things up wrong.  I just tested motor outputs 1-6 on a naze with and without the servo_tilt+camstab option and the mapping vs motors tab is correct.

closed 2016-05-30 by hydra

labomat, 2016-06-10:
I set back the board settings to default and re-entered all settings manually. Now everything works fine...