cleanflight/cleanflight #2163: mixerTricopter

Bengt-M, 2016-05-23:
The default pitch authorities for the three motors on a tri are (4/3, -2/3, -2/3).
The roll authorities are (0, -1, 1).

So the total roll authority is 2 but the total pitch authority is 8/3, which is 33% more. 
Properly balanced, the two front motors are half distance from the CoM, compared to the aft motor. The sum is not much different to a quad plus.

So I would suspect that proper values should be (1, -0.5, -0.5).
(Changing would require new PID tuning).

One consequence is that the pitch PID work in a smaller interval. With an integer PID that means loosing precision, but maybe very little. I don't know if there are other consequences.

Is this an issue or is there a good reason for these values?

hydra, 2016-05-27:
likely the original values stem back from the original multiwii code.  would be good to get some feedback from tricopter pilots.  @stronnag ?

Bengt-M, 2016-05-28:
I don't know how the PID and mixer actually work but my guess is that scaling down these values may only require that the pitch PID coefficients (all 3) should be scaled up accordingly. In this case to multiply them with 4/3. And probably the same idea for some other mixer configurations. Not a big change but giving the PID some more resolution can't be bad. If that is the only effect...

stronnag, 2016-05-28:
I don't think I have the internals knowledge to give an opinion,but if someone prepares a PR (or patch) I'll gladly try it.

Benny-G, 2016-06-13:
Been flying an earlier version of Triflight, been very happy with it. About to update to the latest version to try RC Explorer's new F3 tricopter board. I know there's been talk of merging the branch back in, does Triflight use anything different in this area Bengt?

Bengt-M, 2016-06-13:
Triflight is currently based on Cleanflight v1.12.0 and has not changed these mixer settings.
I have been planning to do some trials in this area using:
```
mixer CUSTOMTRI
mmix reset
mmix 0 1.000 0.000 1.000 0.000
mmix 1 1.000 -1.000 -0.500 0.000
mmix 2 1.000 1.000 -0.500 0.000
```
... and then change the Pitch PID factors accordingly (but I need to do some major repair first).

The big thing with Triflight is how Yaw and Pitch are affected by the servo angle considering both thrust and torque from the aft prop. The improvements in yaw performance are significant. Some users seems to suffer from old Cleanflight bugs related to GPS and other toys that are not main stream for the small racing tricopter users.
The latest beta was released less than a month ago and some more user experience is needed.

Benny-G, 2016-06-13:
Hehe... Had a few "major repair"s myself. The F3 I ordered was to replace the original one I physically broke. Long story. Have one out of 4 naze based tricopters flying right now, I feel your pain. Got the parts for 2 out of the other 3 so wrenching right now and decided to moving to a more current Triflight for all my fleet. 

Have a 230 (trifecta), 280 (rcexplorer mini) a 300 (custom) and a 600 (talon) on various Naze boards. Have a 600 on APM 2.6 as well but I never got into the whole self flying craft thing. GPS is only useful if you want something to fly you a beer cause your too drunk to fish, let alone pilot anything.

Will be on the latest beta when the new FC turns up. (ordered today) Happy to provide feedback but new to GitHub so not sure how to submit it appropriately. Any tips?

Bengt-M, 2016-06-13:
GitHub should be used only for bugs and code discussions. Suggest you start by browsing the user forum on RCExplorer. It's polite to search before you ask (or someone may give you an lmgtfy-link). Best feedback is when users know their setup, can repeat an issue and attach a log.

closed 2017-02-25 by hydra