cleanflight/cleanflight #2165: uncontrollable oscillating throttle with sbus receiver in CF 1.13?

queaker, 2016-05-24:
Hi.
I have a serious problem after upgrading from cleanflight v1.12 to cleanflight v1.13.0 out of the master tree. I flew with the default settings after firmware settings reset. All went ok, but after a few seconds, the throttle input got completly screwed up. See: https://youtu.be/XHcjnrZ6-Uw?t=44s
After second 44 in the video, the copter throttle became completly unresponsive, started by itself and crashed into the building.

The setup is a seriously pro 3 deluxe, frsky x4rsb receiver via sbus on a stock-eachine 250 racer. The receiver worked befor the update in the same cabling. Whats going on here? I hope anyone can help me. Is there a timer issue in conjunction with the BARO mode?

https://www.youtube.com/watch?v=XHcjnrZ6-Uw

[blackbox_log_2016-05-23_202828.TXT](https://github.com/cleanflight/cleanflight/files/278882/blackbox_log_2016-05-23_202828.TXT)

ledvinap, 2016-05-24:
@queaker : Do you remember what happened before throttle increase (in middle of oscillation)? And how did you cut throttle?

ledvinap, 2016-05-24:
Can you also attach cli dump?

Timer issue is possible, but SP3 should have more than enough computing power ...

ledvinap, 2016-05-24:
Seems like usual bug - rcCommand is set on RX rate, but updated on gyroRate ...

`altitudeHold` with `alt_hold_fast_change` enabled or `throttle_correction_value' set will probably cause this behavior  ...

ledvinap, 2016-05-24:
reproduce  altitudeHold + alt_hold_fast_change:

`set alt_hold_fast_change = ON`,  `set rc_smoothing = OFF`, setup BARO mode

connect to configurator,
ARM (with no propellers/unpowered motors),
enter BARO mode (at low throttle),
increase throttle over althold deadband. 

Switch to motor tab. Motor indicators will jump randomly, motors will probably spin up (not tested).

Fast ESC command changes may hurt your ESC ...

ledvinap, 2016-05-24:
same problem with `throttle_correction_value`, but it's not that apparent..

ledvinap, 2016-05-24:
Enabling `set rc_smoothing = ON` will mask this problem, it may be reasonable temporary solution ..

rb1205, 2016-05-25:
I'm having this issue on my naze32 with 1.13 RC1 flashed. Whenever I enable baro mode, motors almost stop and start to twitch, and the quad drops. Blackbox shows that the engine levels jump erratically from 0 to 100 several times in a second.

ledvinap, 2016-05-25:
One nasty property of this bug is that decreasing throttle (below althold threshold) will increase motor power ... 

Just enable `set rc_smoothing = ON` in CLI as workaround ... or wait for proper fix

rb1205, 2016-05-25:
Will do, thanks.

sixsticks, 2016-05-26:
Have same issue with UBAD 30a naze32, DX9. Just flashed with (lastest) Blheli.

andre92, 2016-05-26:
may I see your CLI dump?
mine works fine with stock settings for accelerometer filtering and pids for alt and vel
(naze rev6 acro with sbus)

sixsticks, 2016-05-26:
Sure

sixsticks, 2016-05-26:
> Closed #2165 <https://github.com/cleanflight/cleanflight/issues/2165> via a645e2c <https://github.com/cleanflight/cleanflight/commit/a645e2c69090ee8841bee8224a1c11ce80056514>.

Hi guys,
Is my ‘dump’ still something you can look at? Can’t figure it out need Help! 
Hope this isn’t too much

Sixsticks

# version
# Cleanflight/NAZE 1.12.1 Apr 10 2016 / 19:57:39 (d0c90cf)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature VBAT
feature RX_SERIAL
feature FAILSAFE
feature ONESHOT125

# map
map TAER1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
[... 268 lines cut ...]
set gtune_loP_rll = 10
set gtune_loP_ptch = 10
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

andre92, 2016-05-27:
There is a bug with alt_hold_fast_change. Turn it off and you should be safe. A fix has been merged into master branch yesterday. If you have a compiler you should be able to download and compile the code (make sure you build it for the right target)

atomiclama, 2016-05-27:
Or get it from the nightly builds.
http://cleanflight.memoryleaks.org/builds/cleanflight_cleanflight/

ledvinap, 2016-05-27:
@andre92 , @sixsticks : Using `set rc_smoothing = ON` in CLI is preferred workaround, it should have no negative impact (but some cycles are wasted).

hydra, 2016-05-29:
fix merged in #2176

commit a645e2c6 (2016-05-26): Merge pull request #2171 from andre92/patch-1

Fix #2165, #2149
rcCommand was updated incorrectly
closed 2016-05-29 by hydra