cleanflight/cleanflight #2166: Smart Port  telemetry GPS ground speed unit is not applicable.

hori64, 2016-05-24:
My experience: Speed should be sent in knots/1000.
I repaired and tested. (SPRF3, iNav 1.1, Taranis, OpenTX 2..8)
Please replace in smartport.c this lines:
_uint32_t tmpui = (gpsSol.groundSpeed * 36 + 36 / 2) / 100;
smartPortSendPackage(id, tmpui); // given in 0.1 m/s, provide in KM/H_

This is a fixed and well functioning:
_**//convert to knots: 1cm/s = 0.0194384449 knots
//Speed should be sent in knots/1000 (GPS speed is in cm/s)
uint32_t tmpui = gpsSol.groundSpeed * 1944 / 100; 
smartPortSendPackage(id, tmpui);**_

kgustine, 2016-05-24:
I am also using an SPRF3 w Taranis 9xd plus.  My ground speed shows in meters/sec.  Personally I would rather see it in MPH.  Seems like everyone would want to see different units, so as soon as you change it, others will complain, don't you think?

hori64, 2016-05-24:
OpenTX unit converter works well when it gets this in knots/1000
By default, it converts to knots and it can be changed to any unit of measure.
[image]

ledvinap, 2016-05-24:
As I understand, FrSky specification defined km/h as unit for airspeed and kts for GPS speed. It's complicated for airspeed, but GPS speed is in kts/1000.
https://github.com/opentx/opentx/issues/2305

Displayed unit conversion is transmitter task, not CF.

ledvinap, 2016-05-24:
@hori64 : Can you generate pull request for this change?

hori64, 2016-05-24:
https://github.com/cleanflight/cleanflight/pull/2167
I was correct?!

ledvinap, 2016-05-24:
Thanks, @hori64

closed 2016-06-15 by digitalentity