cleanflight/cleanflight #2169: latest com port driver for OS X

S-T-A, 2016-05-25:
OS X El Capitan Version 10.11.4
where do get a drive to get the com to work???

This was the only solution:
CP210x Macintosh OS X VCP Driver v4 Release Notes
Copyright (C) 2015 Silicon Laboratories, Inc.

This release contains the following components:

* SiLabsUSBDriverDisk.dmg - Image containing the VCP Driver Installer
* ReleaseNotes.txt (this file)
* uninstaller.sh - a bash shell script for removing the driver

Known Issues and Limitations
----------------------------

This release includes the Macintosh OSX driver for the Intel and
PowerPC Platforms versions 10.5, 10.6, 10.7, 10.8, 
10.9, and 10.10.........
--------------------------------

What am i so pose to do?

LaserLock, 2016-05-25:
they work, so, install it

SteveAmor, 2016-05-25:
Hi @S-T-A please do not use the issues log to ask for support and duplicating that support across multiple issues. Please use support forums such as RCGroups for such support. Please close this issue. Thanks.

jflyper, 2016-05-26:
@SteveAmor 
Just for reference, which RCG thread will you recommend for general cleanflight support questions?

JohnOCFII, 2016-05-26:
The thread below is the general RCGroups Cleanflight thread:

http://www.rcgroups.com/forums/showthread.php?t=2249574

jflyper, 2016-05-26:
@JohnOCFII 
I thought it was for F3 targets. At least the title says so...

JohnOCFII, 2016-05-26:
I think the thread title was truncated at some point.  Unfortunately, only the thread owner can change the thread title.  That thread is the main CleanFlight thread.  There is a separate thread for the iNav (GPS, etc.) version, as well as a separate thread for BetaFlight, and a separate thread for RaceFlight (F4 focused).

I'd recommend people read the first 10 or so pages to see the originations of CleanFlight.  The thread started nearly two years ago.  After that, I would read about the last 2 months worth of pages to try to get a sense of where CleanFlight is today.

jflyper, 2016-05-26:
Thanks for the thread history. I didn't know that.

I believe that there are not so many pointers to the RCG thread, even here in the github; the thread url could be posted as a first line of the "Releases" page for example.

JohnOCFII, 2016-05-26:
> I believe that there are not so many pointers to the RCG thread that would guide desparate people, even here in the github. There are many simple improvements, the thread url could be posted as a first line of the "Releases" page, for example.

Good idea.  I don't think I can add persistent text to that page, but I did add the information to the Wiki:

https://github.com/cleanflight/cleanflight/wiki

S-T-A, 2016-05-26:
There is no driver that works on Mac OS x 10.11.5

Unless I'm missing something where do I get a working driver so I can use
to assign the com to work USB to UART??? I have two drones that I need
cleanflight to work correctly to setup my drones....

hydra, 2016-05-27:
if the drivers don't work for a chip, the CP2102, then check the chip manufactuer's website for details and contact them.  We don't write the device drivers, we just use them.

If there are issues then a documentation update would be useful.  Please submit a PR to this effect as required.

closed 2016-05-27 by hydra