cleanflight/cleanflight #2170: Sending arbitrary data from transmitter to FC

brycedjohnson, 2016-05-25:
Would there be use to sending arbitrary data up from a transmitter to the FC?

A couple ideas came to mind to why it would be useful.  Send time up from a taranis to FC so the blackbox logs would be synced up easier.  Sending up GPS to tell it to go somewhere (without have an extra 3dr radio type device like APM uses).  Anyone think of any other reasons something like this would be useful?  Basically it would be an data your transmitter knows, that the FC doesn't, and doesn't need as quick of updates as a regular channel.

I was thinking along the lines of using one of the one of the aux channels.  Dividing the signal into 256 data steps + a couple control bits.

For example if we wanted to send time up to the FC for blackbox.  In this case we'll say the control bits are 
C1 for start of data
C2 for between each data
C3 for End of data
0x01 data is Time

My current time is 5/25/2016, 12:41:19 in epoch time that is 1464198079 secs or 0x5745E3BF in hex.

The transmitter would each chunk of data every 1/10 sec (The speed sent is somewhat arbitrary as long as you have the C2 in there):

C1 0x01 C2 0x57 C2 0x45 C2 0xE3 C2 0xBF C3

Some work on the transmitter side would have to be done as well.  Maybe a lua script on opentx?

Something like that be useful or more confusing and taking up code?

ledvinap, 2016-05-25:
There is serial module for FrSky telemetry. It may be emulated both on CF and OpenTX, providing relatively standard and robust interface.

brycedjohnson, 2016-05-25:
Yes that is very interesting.  I am having trouble finding where people use openTX to send data up the interface (to a FC for example).  Any tips where to look?  Or on the cleanflight side to receive that data?

brycedjohnson, 2016-05-25:
Link to the SP2UART device.  Looks like 30-90 bytes/sec over the air for a bidirectional connection depending on other sensors and such.  

https://f1f0cf2167eff0c12e7f58a76523f833bcf46ea5.googledrive.com/host/0B7dnawamUN6QMWxBNkhzQ0VkYmM/FrSky/SP2UARTmanual.pdf

Anyone know where to find protocol details?

brycedjohnson, 2016-05-25:
Some details here: 
http://openrcforums.com/forum/viewtopic.php?t=4891

And smart port passthrough on opentx
https://github.com/opentx/opentx/pull/3326

ledvinap, 2016-05-25:
Google did not help much ... it seems that non-corrected version sends multiple bytes in 32 bit payload, using different packet IDs to encode packet length.
It should be very easy to reverse-engineer this from communication of SP2UART modules.

I can help decoding the packets if someone can post the dumps (Saleae Analyzer saved data / serial data/ etc)

closed 2017-02-25 by hydra