cleanflight/cleanflight #2175: SP Racing F3 Mini does not respond after setting port GPS

juanmct, 2016-05-28:
Hello.

Board SP Racing F3 Mini. Cleanflight 1.12.
Steps: set up the UART3 port to able GPS, default value in Ports tab, save and reboot, disconnect and remove the USB connection from the FC. 
Afterwards, I tried to connect via USB again with the FC, but that was impossible. Cleanflight message: Failed to connect to port (dont remember exact message). 
I got to reflash firmware with many difficulties again through shorting BOOT pads and tried the GPS  config process again. Same result!

_There is no MicroSD card inserted_

Besides, before this problem showed up, when I selected the Configuration option in Cleanflight, it showed  the loading icon always, never showed the Config data.

I tried to reflash firmware once again but this time with no success. Windows 10 whatever I do (Zadig, etc) cannot recognize the USB and to be honest, the BOOT jumping process is too complicated. I tried pushing both or alone FC buttons but no action. I think one of these buttons should be default-configured to Boot Mode.

I returned the board to the dealer.

juanmct, 2016-06-02:
I'd need to know if this is clearly not an issue with the firmware but instead a faulty board to claim warranty. Thank you

hydra, 2016-06-05:
please use rcgroups for support.  you also double posted this here and via email too.  not cool.

without the board there's not much we can do now, closing.

closed 2016-06-05 by hydra

juanmct, 2016-06-05:
Please check the dates, I sent the email a week later than posting here and asked for help in rcgroups (no answer in there for several days I was checking). As there was no answer in here or rcgroups I sent the email.