cleanflight/cleanflight #2179: 1.13.0-RC1 on SP F3 not recognizing SBUS from Frsky receivers

MeatMuppet, 2016-05-29:
Seriously Pro F3 Deluxe flashed with 1.13.0 does not recognize SBUS from the Frsky X8R or XSR receivers. Changing sbus_inversion had no effect. 

Same configuration of the SP F3 works with Futaba.

Receivers work on a Naze32 flight controller. 

Flashed 1.12.1 back on and the Frsky receivers are now recognized.

Thx.

MeatMuppet, 2016-05-29:
Forgot to mention the Naze32 is flashed with 1.13.0.

MeatMuppet, 2016-05-29:
And I guess I'm not so organized today. Here's a dump of the SP F3 on 1.13.0. This is just after a firmware flash with minimal config to make SBUS work. (verified working on Futaba)

# dump

# version
# Cleanflight/SPRACINGF3 1.13.0 May 12 2016 / 01:55:33 (1e14fd7)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature VBAT
feature RX_SERIAL
feature FAILSAFE
feature BLACKBOX

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
[... 300 lines cut ...]
set gtune_loP_rll = 10
set gtune_loP_ptch = 10
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

gaelj, 2016-05-30:
SBUS on my X8R runs good on a Hobbyking SP F3 Deluxe and on a genuine SP F3 Evo.
Haven't tried it on the Naze. My sbus_inversion is On.
I do have a problem with SmartPort telemetry though. It doesn't work on either board with any combination of CLI setting. Basic telemetry works, I do get the RSSI from the RX but no data from CF.

hydra, 2016-06-10:
just tested this on a SP Racing F3 board here, UART2 for Serial RX SBUS,  UART3 for Spartport Telemetry, works fine.  no inverters, default settings otherwise.  Also tested on SP Racing F3 EVO, works fine there too.

closed 2016-06-10 by hydra