cleanflight/cleanflight #2180: flash cleanflight to cc3d

nitro20, 2016-05-29:
i can get the cleanflight firmware to flash onto the cc3d from libre pilot and then when i open CF and click connect it says "serial port failed to open". tried to flash latest firmware from CF and it pops up that read warning page..i can get it too go back to libre pilot. but i cant get it to connect to CF??

lechak, 2016-05-30:
This is not issue. Please read the docs:
https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20CC3D.md
Section 'Flashing'.
And close issue here.

closed 2016-05-30 by martinbudden

stanleykuenzle, 2016-05-31:
Follow this instructions how to flash the Cleanflight to your CC3D. I've done it yesterday and it worked great.

[``](http://www.southquay3d.com/index.php?route=news/article&news_id=9)