cleanflight/cleanflight #2181: Flip32 (Naze32 rev4) not arming since CF1.12

joehallenbeck, 2016-05-29:
We're having problems arming two Flip32 boards since Cleanflight 1.12
Everything works as expected with 1.11-NAZE, but with 1.12 arming through the RC causes the FC to beep correctly, but throttle doesn't work. The same goes for 1.13RC1 with the exception that a software reset causes all motors to spin without arming them!

Hardware used: EMax MT2204 + EMax mini ESCs, Flip32 10DOF (a.k.a NAZE32 rev.4).
We've experienced these problems on two seperate copters. No problems whatsoever with one using a SPF3Mini and CF 1.12

joehallenbeck, 2016-05-29:
We've got lab equipment locally available for analysis if needed (oscilloscope, function-generator, etc.), but it seems to be a software issue since everything works well after downgrading to CF 1.11

lechak, 2016-05-30:
Maybe you using luxfloat pid controller. Same issue with naze32 rev6
Try search here https://github.com/cleanflight/cleanflight/issues/2091
or here:
https://github.com/borisbstyle/betaflight/issues/110
Or try other pid controllers. MWrewrite work fine

martinbudden, 2016-05-30:
@joehallenbeck 

This is not a bug, it's deliberate behaviour. Cleanflight now has a check to disable arming if the board is overloaded, see:

https://github.com/cleanflight/cleanflight/blob/master/src/main/mw.c#L263

Previously the board could arm, but this could then cause problems in flight.

martinbudden, 2016-05-30:
Duplicate of https://github.com/cleanflight/cleanflight/issues/2091

closed 2017-02-25 by hydra