cleanflight/cleanflight #2186: Is gtune working?

limhyon, 2016-06-01:
Hi,
I've tried to test gtune feature with my Naze rev5 (most recent stable version 1.12.1).
What I read from code is
 - Set gtune mode by assigning AUX switch
 - If I set all pid gains to zero, then it will change to 1.0, immediately after I change to gtune mode with arm

However, it doesn't changed. So I've tried old firmware posted in
http://www.rcgroups.com/forums/showpost.php?p=31525114&postcount=7150

It does change to 1.0 immediately after I change the flight mode to GTUNE with ARMED.

MJ666, 2016-06-01:
It was never recomended to start with zero gain on any PID values for G-Tune? You shoul start with the defaults. G-Tune is only changing P value. 1.0 is the default lower limit for tuning and this can be changed with the related CLI settings. 

https://github.com/cleanflight/cleanflight/blob/master/docs/Gtune.md

Not sure how it was affected by the recent scaling changes. Since we have black box today and I have more experiance with tuning I don't rely on G-Tune anymore and have not used this for an longer time. G-Tune was some help but not the solution for an well tuned copter.

bcdebusk, 2016-06-01:
I'm on the latest version of Cleanflight for the Naze32 Rev6.  I'm starting with the default PIDs and the vehicle runs fairly well.

- Programmed a switch for GTUNE (and switch correctly turned green when activated)
- Used Acro mode only
- Set a deadzone for AER to 6 (to avoid jitter stopping the algorithm)

I then flew around with minimal corrections of the craft but beforehand, I tried every combination:

- Power / Arm / Gtune On / Fly
- Power / Gtune On / Arm / Fly

And I can't make it do the first twitch.  But just in case it was being subtle, I tried to save the would-be values.  So after flying, I then tried:

- Gtune Off / Land / Disarm / Gtune On
- Land / Gtune Off / Disarm / Gtune On

And I can't get it to do anything.  Is there a glitch in GTUNE for the Naze32?

Thanks in advance,

Brian

MJ666, 2016-06-01:
Wich version you are flying.There was an bug introduced a while ago wich was fixed recently. Try with one of the latest RC builds.

bcdebusk, 2016-06-01:
I'll do that.  I haven't been looking at RC builds.

Brian

bcdebusk, 2016-06-02:
I loaded the latest release candidate last night.  I did one tuning session and it did appear to adjust the P values slightly.  It didn't adjust the D or I terms at all -- is there a reason behind that?  All in all, the quad flew well.

Old P-values:  4.0, 4.0, 8.5
New P-values: 4.9, 4.2, 10.0

So I tried another tuning session with the same copter and re-saved the values.  This time, it set all the pitch and roll P values to 10 and the yaw P value to 1.  That made no sense.

Any ideas or thoughts on this?

bcdebusk, 2016-06-03:
Did another tuning session.

Original P-values: 4.9, 4.2, 8.5
New P-values: 9.6, 10.0, 1.0

So it tried to essentially put everything back to 10, 10, 1.

Is there any possible way a 3S QAV250 would need P-values that high?

hydra, 2016-06-05:
we cannot say, pid values are unique to every build.

to answer your original question 'is gtune working'  yes is the answer, nothing more to do in this issue.  if you find bugs or have code improvements to offer please let us know.

closed 2016-06-05 by hydra