cleanflight/cleanflight #2191: Consider code style changes

hydra, 2016-06-01:
We should conside new code style for 1.14 and onwards.

Here's one:

For a long time I've not liked the style:

```
void someFunction() 
{
}
```

i prefer

```
void someFunction() {
}
```

I'm sure @ledvinap @martinbudden and @digitalentity will want to suggest other changes, let's hear them!

gaelj, 2016-06-01:
Hi @hydra, I am all in favor of standardisation and I have no problem with adhering to whatever coding style is choosed.
Just to clarify though, don't you mean that you prefer :
```
void function()
{
}
```
over 

```
void function() {
}
```

?
As that's the new way it's coded in your referenced commit.

I have a few suggestions regarding coding style :

- `{ }` should generally not be used around 1-liners (but can be tolerated for code clearness, for example when inside multiple nested blocks).
- In an array definition, the contents should be separated from the brackets by a space.
Example: `{ 2, 50 }` rather than `{2, 50}`.
- a function return should be enclosed in parentheses. For example `return foo;` rather than `return (foo);`.
- Enforce using parentheses around each group in logical and mathematical statements, rather than rely on the compilator's logic which usually produces slightlky shorter, but less clear code, and in the worse case can yield unexpected results when the compilar does not behave as expected.
- For the CFC some extra JS / HTML / CSS rules could be defined. For example, is it prefered to use single ou double quotes for string literals ?

ledvinap, 2016-06-01:
Let's start with https://www.kernel.org/doc/Documentation/CodingStyle and add (as few as possible) exceptions ...

martinbudden, 2016-06-01:
Well, over the years I've probably worked on C and C++ projects that used all the variations of placement of braces, so I'm used to going with whatever the house style is. In terms of the standardisation of whitespace I favour styles that keep the code compact - I find it easier to code if I can see more lines of code on the screen. So from that point of view: 

(1) I prefer it if open brace characters are not on a line of their own, e.g.

```
if (condition) {
    statement;
}
```
rather than
```
if (condition)
{
    statement;
}
```

(2) In terms of conditional statements and loops I think `{ }` should **always** be used. The reason for this is simple: if a PR turns a one-liner into a two-liner then the diff will contain 2 changed lines and an additional line, rather than just an additional line - this makes the diff more difficult to review.

(3) For discussions, points should be numbered rather than bulleted ( @gaelj :-) ) - this means that discussion points can be more easily referred to.

(4) @gaelj , your point (3) is self-contradictory.

I'll add further statements in a further comment.

martinbudden, 2016-06-01:
This comment contains my views on coding style not related to formatting.

1) We should take maximum possible advantage of compile time checking, so generally warnings should be as strict as possible.
2) Variables should be declared at point of usage. (Generally this makes code changes more local, so makes reviewing PRs easier).
3) Const-correctness should be enforced.
4) Data should move from right to left, e.g.
```memcpy (void * dst, const void * src, size_t size)``` not ```memcpy (const void * src, void * dst, size_t size)```
This also mimics the assignment operator ( e.g. ```dst = src;```)
5) When a group of functions act on an 'object' then that object should be the first parameter for all the functions, e.g.:
```
float applyBiQuadFilter(biquad_t *state, float sample);
void BiQuadNewLpf(biquad_t *state, float filterCutFreq,uint32_t refreshRate);
```
rather than
```
float applyBiQuadFilter(float sample, biquad_t *state)
void BiQuadNewLpf(float filterCutFreq, biquad_t *state, uint32_t refreshRate)
```

unitware, 2016-06-22:
I'd prefer that all files include their own dependencies so that I do not get compiler errors if I include io/serial.h but not drivers/serial.h for example.

And the style should be possible to enforce with a tool like astyle or similar, possibly included in the build as a separate test.

tannewt, 2016-06-22:
Formatting and include policy is easy. Use clang-format and include what
you use to have the computer redo it for us. Links for those are
http://clang.llvm.org/docs/ClangFormat.html and
http://include-what-you-use.org/ respectively. It's easiest to enforce
because it's automatic.

I used to be at Google and that's their policy. Here's a good talk about
over style decisions http://youtu.be/NOCElcMcFik.

martinbudden, 2016-06-23:
I'm in favour of include what you use, especially if it can be done using a tool. It makes the code easier to understand and it also improves build times. The latter is becoming significant with Cleanflight, build times are already a disincentive to making all targets before a PR (just let the travis build do it for you), but it is also becoming an issue with the travis build, since at critical times (i.e. as a release approaches) the travis build queue gets clogged up.

hydra, 2016-07-02:
@ledvinal said:

code:
`typedef void (*handlerFuncPtr)(void);`

`typedef void handlerFunc(void);` is easier to read

I noted that the code in question was following current pattern, all occurrences should be fixed.

Bengt-M, 2016-07-31:
Interesting as well as surprising to see this detailed discussion about formatting. I thought that was settled long ago.

[Development.md](https://github.com/cleanflight/cleanflight/blob/master/docs/development/Development.md) points at [CodingStyle](https://github.com/multiwii/baseflight/wiki/CodingStyle) which in turn points at K&R. 

Maybe that CodingStyle document should be replaced/updated but K&R are usually right and can be enforced by most tools.
I have done some PR's in another fork and always try to maintain the style of the original author, but when it's not K&R I find it slightly more difficult.

hydra, 2016-08-23:
Loads of very good stuff in here.

Can someone update the documentation with the content of all the good stuff in here?  No need to schedule this for the 1.14.0 release but we should look to do it when we have time, as always there are bigger fish to fry but at least we can point people to this discussion for now.

Bengt-M, 2016-09-14:
So now after b39fc74 I suppose this can be closed.

mkschreder, 2016-10-24:
I think all of these things are pretty minor. It is more important to keep code portable and to code in a more "object oriented" manner. 

ie: 
struct myobj {
   int somevar; 
}; 

instead of: 
static int somevar; 

and for methods that work on objects the code should be: 
myobj_method(struct myobj *self){
    self->someval = whatever; 
}

Note that the above code is reentrant while the version using static global variable is not. Maintaining reentrancy is a good goal generally because aiming for it helps a lot in keeping the code clean and with minimal external dependencies.

closed 2017-02-25 by hydra