cleanflight/cleanflight #2192: Unable to arm, NAZE32 V5 Cleanflight 1.13 RC2

regnets, 2016-06-01:
Hi,

today i tried RC2, however i am unable to arm, normally i arm with AUX1, after this didn't work, i tried to arm with the left stick (MODE 2). Doesn't work either.

Here is the dump of my settings:
`Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# Cleanflight/NAZE 1.13.0 May 30 2016 / 22:43:35 (0245144)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature VBAT
feature RX_SERIAL
feature FAILSAFE
feature LED_STRIP

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 4,15::A:0
led 1 5,15::A:0
led 2 6,15::A:0
led 3 7,15::A:0
led 4 8,15::A:0
led 5 9,15::A:0
led 6 10,15::A:0
led 7 11,15::A:0
led 8 0,0:::0
led 9 0,0:::0
led 10 0,0:::0
led 11 0,0:::0
led 12 0,0:::0
led 13 0,0:::0
led 14 0,0:::0
led 15 0,0:::0
led 16 0,0:::0
led 17 0,0:::0
led 18 0,0:::0
led 19 0,0:::0
led 20 0,0:::0
led 21 0,0:::0
led 22 0,0:::0
led 23 0,0:::0
led 24 0,0:::0
led 25 0,0:::0
led 26 0,0:::0
led 27 0,0:::0
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

# mode_color
mode_color 0 0 1
mode_color 0 1 11
mode_color 0 2 2
mode_color 0 3 13
mode_color 0 4 10
mode_color 0 5 3
mode_color 1 0 5
mode_color 1 1 11
mode_color 1 2 3
mode_color 1 3 13
mode_color 1 4 10
mode_color 1 5 3
mode_color 2 0 10
mode_color 2 1 11
mode_color 2 2 4
mode_color 2 3 13
mode_color 2 4 10
mode_color 2 5 3
mode_color 3 0 8
mode_color 3 1 11
mode_color 3 2 4
mode_color 3 3 13
mode_color 3 4 10
mode_color 3 5 3
mode_color 4 0 7
mode_color 4 1 11
mode_color 4 2 3
mode_color 4 3 13
mode_color 4 4 10
mode_color 4 5 3
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3
mode_color 6 0 6
mode_color 6 1 10
mode_color 6 2 1
mode_color 6 3 0
mode_color 6 4 0
mode_color 6 5 2
mode_color 6 6 3
mode_color 6 7 6

set looptime = 2000
set emf_avoidance = OFF
set i2c_highspeed = ON
set gyro_sync = ON
set gyro_sync_denom = 1
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rssi_ppm_invert = OFF
set rc_smoothing = OFF
set rx_min_usec = 885
set rx_max_usec = 2115
set serialrx_provider = IBUS
set sbus_inversion = ON
set spektrum_sat_bind = 0
set input_filtering_mode = OFF
set min_throttle = 1150
set max_throttle = 1850
set min_command = 1000
set servo_center_pulse = 1500
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set retarded_arm = OFF
set disarm_kill_switch = OFF
set auto_disarm_delay = 0
set max_arm_angle = 25
set small_angle = 25
set fixedwing_althold_dir = 1
set reboot_character = 82
set gps_provider = NMEA
set gps_sbas_mode = AUTO
set gps_auto_config = ON
set gps_auto_baud = OFF
set telemetry_switch = OFF
set telemetry_inversion = OFF
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = OFF
set current_meter_type = ADC
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = DEFAULT
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set max_angle_inclination = 500
set gyro_lpf = 188HZ
set gyro_soft_lpf = 60
set moron_threshold = 32
set imu_dcm_kp = 2500
set imu_dcm_ki = 0
set pid_at_min_throttle = ON
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = ON
set servo_lowpass_freq =  400.000
set servo_lowpass_enable = OFF
set failsafe_delay = 10
set failsafe_off_delay = 200
set failsafe_throttle = 1000
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set failsafe_procedure = 0
set acc_hardware = 0
set baro_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = SERIAL
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0

# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h

# dump profile

# profile
profile 0

# aux
aux 0 0 0 900 900
aux 1 0 0 900 900
aux 2 0 0 900 900
aux 3 0 0 900 900
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

# rxrange
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000

# servo
servo 0 1000 2000 1500 45 45 100 -1
servo 1 1000 2000 1500 45 45 100 -1
servo 2 1000 2000 1500 45 45 100 -1
servo 3 1000 2000 1500 45 45 100 -1
servo 4 1000 2000 1500 45 45 100 -1
servo 5 1000 2000 1500 45 45 100 -1
servo 6 1000 2000 1500 45 45 100 -1
servo 7 1000 2000 1500 45 45 100 -1

set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83
set gps_wp_radius = 200
set nav_controls_heading = ON
set nav_speed_min = 100
set nav_speed_max = 300
set nav_slew_rate = 30
set alt_hold_deadband = 40
set alt_hold_fast_change = ON
set deadband = 0
set yaw_deadband = 0
set yaw_control_direction = 1
set 3d_deadband_throttle = 50
set throttle_correction_value = 0
set throttle_correction_angle = 800
set default_rate_profile = 0
set gimbal_mode = NORMAL
set acc_cut_hz = 15
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff =  5.000
set acc_unarmedcal = ON
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf =  0.600
set baro_cf_vel =  0.985
set baro_cf_alt =  0.965
set mag_declination = 0
set pid_controller = MWREWRITE
set p_pitch = 40
set i_pitch = 30
set d_pitch = 23
set p_roll = 40
set i_roll = 30
set d_roll = 23
set p_yaw = 85
set i_yaw = 45
set d_yaw = 0
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 20
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1
set yaw_p_limit = 500
set dterm_cut_hz = 0
set gtune_loP_rll = 10
set gtune_loP_ptch = 10
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

# `

regnets, 2016-06-01:
Just gone back to 1.12.1, everything is working fine.

Now back on 1.13 RC2, i am unable to arm.

regnets, 2016-06-01:
Are you sure i got GPS/Baro enabled? 

I didn't activate it intentionally.

Should the default configuration working on all boards? I saw that the looptime setting was to 2000 (which is fine i think).

Is it possible to see the current load? I am sure that my board got 1000 cycles, so the load should be fine?

It is a pitty that you are not able to transfer your settings from 1.12.1 to 13, so i have to setup everything again for the test.

I will come back to this issue at the weekend if this is ok for you?

gaelj, 2016-06-01:
The `set gyro_sync_denom = 2` workaround is now part of the code for the boards where it was needed, so in theory it shouldn't change anything:
```
#ifdef STM32F10X
    // avoid overloading the CPU on F1 targets when using gyro sync and GPS.
    if (imuConfig()->gyroSync && imuConfig()->gyroSyncDenominator < 2 && featureConfigured(FEATURE_GPS)) {
        imuConfig()->gyroSyncDenominator = 2;
    }
#endif
```

CLI command `status` gives you the system load.

swilkens, 2016-06-11:
Hi again,

I dug my old naze rev 5 quad out of the dust and did some checking with 1.13 FINAL.

With baro set to auto (which is what you have), my controller shows an average load of ~0.75

If I set the baro to forced off using `set baro_hardware = 1`, that average load drops to about ~0.35

if I then also set the mag to forced off using `set mag_hardware = 1`, the average load drops to about ~0.15

It seems you can remove quite a bit of CPU load if you forcibly disable hardware you're not using. 

The baro is safe to force off if you're not using it for altitude hold. I'm not sure if the mag is currently used outside of heading hold, though. Somebody else might be able to tell you if that's safe to force off as well - assuming you're not using heading hold mode.

juanmct, 2016-06-11:
Hi, 

same issue with a Naze32 Rev6 10DOF. Had to disable the barometer to be able to arm. I noticed the system load can vary with the same configuration, so could be the case to arm from time to time?

In addition, could be added a parameter to handle this system max load value from CLI so each pilot would be the responsible of this overload risks?

```
# status
System Uptime: 49 seconds, Voltage: 122 * 0.1V (3S battery - OK), System load: 0.80
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=MS5611
Cycle Time: 1002, I2C Errors: 0, registry size: 46

# status
System Uptime: 51 seconds, Voltage: 122 * 0.1V (3S battery - OK), System load: 0.72
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=MS5611
Cycle Time: 999, I2C Errors: 0, registry size: 46

# status
System Uptime: 56 seconds, Voltage: 122 * 0.1V (3S battery - OK), System load: 0.71
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=MS5611
Cycle Time: 998, I2C Errors: 0, registry size: 46

# status
System Uptime: 57 seconds, Voltage: 122 * 0.1V (3S battery - OK), System load: 0.67
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=MS5611
Cycle Time: 997, I2C Errors: 0, registry size: 46

# status
System Uptime: 58 seconds, Voltage: 122 * 0.1V (3S battery - OK), System load: 0.67
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=MS5611
Cycle Time: 991, I2C Errors: 0, registry size: 46

# status
System Uptime: 60 seconds, Voltage: 122 * 0.1V (3S battery - OK), System load: 0.69
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=MS5611
Cycle Time: 1002, I2C Errors: 0, registry size: 46

# status
System Uptime: 61 seconds, Voltage: 122 * 0.1V (3S battery - OK), System load: 0.79
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=MS5611
Cycle Time: 997, I2C Errors: 0, registry size: 46
```

graphyphoto, 2016-06-14:
Naze32 Rev6 won't arm on 1.13 with LED_STRIP enabled. If disabled, the Naze32 will arm. Switching to 1.12, the Naze32 will arm with LED_STRIP enabled.

regnets, 2016-06-15:
That could be also my problem. I had the led strip feature enabled (so that i can easily see that the copter is armed).

swilkens, 2016-06-18:
@graphyphoto @regnets did you guys check the CPU load when you attempted an arm?

I just enabled LED_STRIP on my rev 5 running 1.13 final and had no issues arming the board.

regnets, 2016-06-28:
@swilkens do you got the 6DOF or 10DOF version? I am still unable to arm with just the led strip feature enabled.

closed 2017-02-25 by hydra