cleanflight/cleanflight #2202: Air mode causing issues.

jamespwright, 2016-06-02:
I've been flying my quad for just under 2 months and it flies great except for a small problem of drift at 0 throttle.  I was told to enable Air mode to fix this.  Air mode works great to fix that issue, however Air mode has introduced random losses of control that do NOT happen if Air mode is off.  

The "random losses of control" seem to occur in two ways:
1) The quad gives itself a quick "shake".  Think "invisible dog grabbed it and shook it really quickly 1-2 times".  After the shake is over it flies normally again.  
2) The quad tilts sideways and doesn't respond to any controls.  Then it hits the ground.  

I've tried Air mode 4 times now and each of these issues has repeated at random.  Sometimes I'm flying for 5 minutes with no problems, other times it happens within 10 seconds of takeoff.  

I don't have a blackbox and I don't have video.  What information can I provide you with to help debug this?

I'm running a Naze32 (Rev 5 I think).  DYS 20A ESCs, GoolRC 2204, 2300kv motors.  What else do you need to know?  Here is a dump of my settings: https://gist.github.com/jamespwright/8feb4bff888c4ba15f7c896af76ce5bd

borisbstyle, 2016-06-03:
Airmode is a bit more agressive to ESC's. The throttle changes can be much faster and some ESC's can be pushed over the edge earlier.
Sometimes it helps raising the min_throttle higher to prevent ESC's to stall due to prop wash air.

hydra, 2016-06-05:
yeah, likely just a configuration issue.  for support please use rcgroups, if you confirm it's not a hardware or configuration issue please give details.

closed 2016-06-05 by hydra

jamespwright, 2016-06-06:
For now it seems to have been a min_throttle issue.  It was set too low so one motor was possibly activating before the other 3.  I have flown only 5 batteries since making the change but haven't had any issues so for now I'll close this as a non-issue.