cleanflight/cleanflight #2205: mode_color CLI command is not recognized

rigo3t, 2016-06-03:
Hi there, 

I have been trying to change the LED color for different modes, for armed and for disarmed. According to the instruction PDF the way to do this is using the command mode_color and 3 arguments. When I insert this "mode_color 6 0 4" I get this: 

# mode_color 6 0 4
Unknown command, try 'help'

Is there something else I should do? I have the LED strip enabled, everything else is working.

rigo3t, 2016-06-03:
Sorry, I realize I probably should have stated what I'm using. This is on a Afroflight Naze 32 Rev 6. I'm using an addressable LED panel with 6 LEDs. 

Here are examples: According to the manual, if I just type mode_color it should list the modes and their corresponding colors.

[image]

set mode_color gives me an "Invalid name"

[image]

gaelj, 2016-06-03:
Hi, it's a new feature. Try the unstable release 1.13 RC2.

rigo3t, 2016-06-03:
Ohh! I didn't realize that, my bad. Thanks for the headsup. Quick question... In terms of flight characteristics, how stable is this release? I have a brand new build and I want to be able to fly it this weekend without major issues. Any thoughts?

gaelj, 2016-06-03:
1.13 RC2 flies good for me, better than 1.12 in my opinion. Lots of bug fixes & some new features as well (check the release notes & git diff). The bugs reported in RC1 have been fixed.
So I don't think you are taking a big risk using RC2.
Additionally, if you do find any bug, report it and you will have contributed in making 1.13.0 even better.

rigo3t, 2016-06-03:
Fair enough, I'll give it a spin. Thanks Gael!

gaelj, 2016-06-03:
Can you close the issue please ?

rigo3t, 2016-06-03:
Absolutely. Thanks!

closed 2016-06-03 by rigo3t