cleanflight/cleanflight #2206: Add some new PWM protocols (Multishot and Oneshot42)

borisbstyle, 2016-06-03:
A lot of new ESC's around with new PWM protocols so have to support that.

ledvinap, 2016-06-03:
@borisbstyle : Is there any advantage?
IMO 1kHz loop time is enough / a bit more  than necessary. And OS125 is adequate. Reducing pulse time will increase jitter/noise/temperature influence ..

borisbstyle, 2016-06-03:
@ledvinap 
I would pretty much agree with you, but I just see a trend where code would need to adapt to. Besides that it is very easy to implement anyway.
Some say that ESC's do averaging of last few pulses and that faster refresh rate decreases delay significantly just because you can manipulate the average on that way. This should be solved on the ESC side anyway.

ledvinap, 2016-06-03:
@borisbstyle : Implementing it is not that simple - oneshot pulse is sent once per loop, sending shorter pulse will have not increase pulse frequency (no improvement in averaging filter delay), only negative effects will apply.
And is ESC update should be decoupled from loop, then it's easier to use PWM

borisbstyle, 2016-06-03:
No it will not, but it can also be done unsynced just like regular PWM as
well with configurable motor pwm rate.

But even synced shorter pulse means less delay

ledvinap, 2016-06-03:
Unsynced is equal to PWM (possibly with narrower min/max limits). It will probably need different prescaller, complicating coexistence with other functions.
When synced, OS125 delay is 250us max. Even averaging 2 samples in ESC will introduce 500uS delay

Current ESC interfaces are quite well-behaved - it may be possible to implement OS125 and slower PWM modes using interrupts (planing next compare match dynamically), thus making it easy to share timer for different functions (counter will always count to 0xffff, only prescaler will possibly change). Implementing these fast modes will complicate it in future.

My personal opinion is that OS42/MS almost complete nonsense. Someone discovered HW input capture and is trying to push it as far as possible, just for the sake of it. Implementing this in CF will give credibility to this whole idea. And it may easily become support nightmare ...
Digital ESC interface is what is needed. Just use pulse-position/pulse length coding to encode 4-5 bits per pulse and use few pulses per ESC command (at manageable pulse length/frequency comparable to OS125).

borisbstyle, 2016-06-03:
Well I totally agree about the digital protocol what would be much better
without doubt.

But yeah you are probably right about it. Lets park it than for a while

hydra, 2016-06-05:
I am in agreement with @ledvinap we *really* don't want to make problems for ourselves that we don't need.  I think we have bigger fish to fry at the moment too.  Once the dust has settled we could look but as of right now and long term I'm in favour of a proper digital solution.

closed 2017-02-25 by hydra