cleanflight/cleanflight #2211: Prepare for 1.14.0 release

hydra, 2016-06-05:
TODO
#2210 OSD Support (If existing maintainers/contributors want free hardware let me know!)
#2014 Configuration menus.
Betaflight mergebacks. (@martinbuddon can you chat to boris - there have been some exciting developments in the last week with regards to the pilot feel improvements).

NOTES
I want 1.14.0 out the door quickly.

I'm looking for a much shorter release cycle for 1.14.0 and a tighter focus.

FUTURE
The release 1.15.0 will focus on #884 #1984 - IO allocation.

martinbudden, 2016-06-06:
@hydra , WRT betaflight mergebacks - I've been following @borisbstyle changes and have some PRs on the way.

jflyper, 2016-06-07:
Isn't it about time for:
Feature exti 2 #1984 
Implementation of serial passthrough. #1620

Oh, and of course,
Code from jflyper. Fixes issue 1868 #1907

#1868 is actually a terrible bug that refers unrelated memory in stack.
(and syntactically incorrect, too, intermixing trigger and echo.)

martinbudden, 2016-06-07:
@hydra , I welcome shorter release cycles with tighter focus. I think this is a good thing in general, not just for 1.14

However, just because a release is focused on one feature, that does not mean all small changes should be deferred to the next release. If that release is also focused then the small changes would be further deferred etc. There needs to be a balance.

I'll give two examples to make my point:

1. PR https://github.com/cleanflight/cleanflight/pull/2157 , which is just a spelling correction
2. PR https://github.com/cleanflight/cleanflight/pull/1907, mentioned above, this is a simple bug fix.

These sorts of changes should simply be merged in at the beginning of the release cycle.

SteveAmor, 2016-06-08:
I agree as PR #1052 will likely never be merged. ;-)

hydra, 2016-06-10:
@jflyper yes #1984 is next after v1.14.0

hydra, 2016-06-10:
@SteveAmor scheduled #1052 for v1.14.0, thanks for the reminder.

hydra, 2016-06-10:
@martinbudden Yes, 'quick wins' are fine.  Agreed regarding balance, but if you try to do too many 'quick wins' then it doesn't become quick at all, and the amount of changes add up and complicate testing and release confidence.

hydra, 2016-08-10:
Ok, much OSD work has been done but I'm going to park the remaining OSD code for now so we can get a new version out the door ASAP.

Things have been very slow due to lack of available time, I'm working with people to try and resolve this situation in the future.  Turns out that finding a house to buy and live in and moving country is somewhat time consuming...

hydra, 2016-08-11:
@SteveAmor #1052 merged.
@jflyper #1984 merged.

hydra, 2016-08-11:
@martinbudden how are the betaflight mergebacks coming along?

martinbudden, 2016-08-11:
@hydra , I've added link in task list to betaflights mergebacks issue (which is itself a task list).

As you can see, I did some work on that a while ago, which has been pulled in. Realistically I'm not going to be able to do any more before the next Cleanflight release.

Also see my comment at https://github.com/cleanflight/cleanflight/issues/2213#issuecomment-238981673 - there have been huge changes in this area in betaflight. You need to think about whether you want to cherry pick them, or take them wholesale. Either way any more mergebacks should probably be delayed until the next release.

hydra, 2016-08-23:
ok, #2014 is deferred due to lack of time.

hydra, 2016-08-24:
I went through and either deferred or closed some other tasks, everything in the 1.14.0 milestone is what needs to be done, no more, no less.

hydra, 2016-08-29:
We should also merge the bug fixes from iNav issue https://github.com/iNavFlight/inav/issues/431

- [ ] handle compas initialisationread failures
- [ ] delay before attempting to initialise gps (wait for GPS boot)

martinbudden, 2016-08-29:
@hydra , I'll make a PR for the iNav issues.

hydra, 2016-08-29:
@martinbudden awesome, thanks very much.  In related news I posted out an FC to you today.  There was a couple of days where I didn't get to the post office last week due to other commitments.

martinbudden, 2016-08-29:
Cleanflight ULBOX buffer size `#define MAX_UBLOX_PAYLOAD_SIZE 344` -  that's bigger than iNav, which is now 256 after size increase. (iNav has buffer overflow protection.)

martinbudden, 2016-08-29:
@hydra , all PRs done. You may wish to cherry pick the GPS delay, as discussed in the PR.

hydra, 2016-08-30:
IIRC I added the buffer overflow protection in the UBLOX code back when I was looking for possible reasons for intermittent GPS behavior issues.  Need to make sure we have it still.

digitalentity, 2016-08-30:
@hydra just checked, the UBLOX buffer overflow is still there.

hydra, 2016-09-02:
@digitalentity good to know, any chance of a PR for that?

I've been out testing new VTX and OSD code and haven't had the chance to look at this issue much recently.

I want to get 1.14.0-RC1 out by the the end of next week. (11th September)

digitalentity, 2016-09-02:
PR for what? GPS buffer overflow protection code is already in Cleanflight
:)

hydra, 2016-09-02:
Ok i mis-understood your comment then :smile:  great!

hydra, 2016-09-02:
@martinbudden I merged the other PR's for the issues that were found recently, many thanks for your contributions as always.

hydra, 2016-09-02:
I will work on the battery cleanup and msp changes so we can get this out there next week.

There is still work to do on the OSD support but some of that will come later.  I want a new stable version out ASAP, we can extra bells/whistles in due course.

hydra, 2016-09-11:
#2396 (battery cleanup) is now in progress - partially tested and working on the SPRacingF3EVO and the SPRacingF3OSD.

hydra, 2016-09-12:
I opened issue #2395 - i found during testing.

hydra, 2016-09-12:
ok, #2396 is done and merged.

I will update the configurator to match the new code, probably tomorrow now as I ran out of time today.  Maybe there will be a minor fix or two but the main re-organization/refactoring work and support of additional voltage sources throughout the system is now done.

hydra, 2016-09-13:
ok, i tagged and pushed v1.14.0-RC1 binary up the releases page on github.
for SPRacingF3OSD users only.
RC2 will have FC binaries once issue #2395 is addressed.

hydra, 2016-09-14:
Perhaps controversially i merged in #2364 Horizon Mode SAFE/EXPERT modes.

hydra, 2016-09-14:
NAZE target is currently too big for the build to pass now.  I feel a code-cull coming on...

martinbudden, 2016-09-15:
What version of the compiler are you using? I'm using 5.4 and it builds fine on my machine.

Getting rid of GTUNE saves about 1.5K.

hydra, 2016-09-15:
it was a build failure on travis and on my machine here.  5.4 too.

hydra, 2016-09-15:
@martinbudden so other than the PG LIST/GET/SET via MSP it seems there is nothing else to do!

However, it's not very exciting unless you have an OSD board :smile: 

I was thinking about merging some of the tried and tested stuff from the other forks and some other PR's before we do a release candidate for the FC's.

The Notch filter in betaflight looks like it would be useful.  And perhaps some of the control changes.

Thoughts?

martinbudden, 2016-09-15:
Well, my instinct would be to release what you have as soon as possible and say it is a release to support your OSD board. People who buy one of your boards are going to want something to run on them.

Then target a new release for one month's time and put all the tried and tested stuff in that release.

The notch filter does indeed look good, but implementing that means changes to the configurator and also sufficient testing on a variety of models to set a good default value. If you rush it you will be compared unfavourably with betaflight - better to make a decent job of it.

hydra, 2016-09-17:
good feedback martin.  I don't see the harm however in merging some of the features and doing a a couple of release candidates, that's the phase where we can do our flight testing and get feedback.

There's already the new horizon mode changes which we also need to get additional testing and feedback on too so I think porting a few things over will be OK.  Certainly I'm not going to release it if it's not ready.

hydra, 2016-09-23:
Ok, I've decided we *should* merge the BF #2416 in for the 1.14.0 release.  I want the IO code changes ASAP but we *need* better filtering for those suffering from vibration issues to try.  The latter is more important for the users.  After spending a bit of time with the #2416 code, extracting various bits and merging them to master and also flying the remaining code still in the PR I'm fairly happy with it.  I will be doing more code cleanups, more flight testing and making sure the defaults are OK on all the boards we support next week.

hydra, 2016-10-13:
OK, #2416 is merged.  I want IBus telemetry too and it looks quite isolated.

Can those with iBus equipment please test-fly #2415 and provide feedback on the PR?

unitware, 2016-10-22:
@hydra et. al 
Test flight today with v1.14.0-RC2 looks stable with the Ibus telemetry stuff working.

(No advanced flying, just hovering and cruising around in the garden in horizon and acro mode).
My previous stability problems came from using propguards and 0630 props, now without propguards stuff is more stable using default PID settings)

kitchung, 2016-10-24:
The doc "Board - SPRacingF3Evo.md" says "Sonar will be supported in CF 1.14" for the SP Racing F3 Evo. Is this still going to happen with 1.14? I didn't see any mention in the release notes for both RC1 and RC2.

hydra, 2016-10-30:
@kitchung done via 6b03f97c522442b27b3c480ab79a2d85ded85366

Please test a new development build that includes that commit.

Thanks for the reminder too.

I think CF 1.14.0 is almost done.

kitchung, 2016-10-30:
Thanks! Unfortunately I do not have any 1k resistors on-hand and will arrive some time next week.

kitchung, 2016-10-30:
FYI, need to uppdate https://github.com/cleanflight/cleanflight/blob/master/docs/Sonar.md to note support.

kitchung, 2016-11-07:
FYI bench test of sonar on SP F3 EVO with Clean flight 1.14 was successful.

closed 2016-11-17 by hydra