cleanflight/cleanflight #2213: Betaflight mergebacks

martinbudden, 2016-06-06:
Merge selected betaflight improvements into cleanflight

1. Use PT1 filters rather than biQuad filters for PIDs. Improve filter defaults.
2. Use calculation rather than lookup for RC curves
3. Add super expo feature
4. Add Dynamic PIDs (P accelerator)

martinbudden, 2016-06-06:
@borisbstyle , @hydra has asked me to look into betaflight "pilot feel" improvements with a view to merging them back into Cleanflight. I've listed the main ones I'm aware of above - are there any others you think should be merged into Cleanflight.

My plan is to submit a series of PRs (rather than one big PR) to get these merged back in.

borisbstyle, 2016-06-06:
Cool @martinbudden. I told that to dominic. I dont have much time to merge back a lot as I still hope to make some progress.
I will help you out.

Dont merge dynamic P yet. I am not yet happy with it in some scenarios.

martinbudden, 2016-06-06:
@borisbstyle thanks.

lkaino, 2016-06-07:
I would add VCP blheli passthrough to the list.

ledvinap, 2016-06-07:
@lkaino BLHeli 4-way is supported in CF ...

lkaino, 2016-06-07:
Ah, sorry for my ignorance. Couldn't connect 4-way with the VCP passthrough. All good then :).

hydra, 2016-06-09:
unrelated comments deleted.

martinbudden, 2016-06-11:
Dynaminc P removed from todo list.

hydra, 2016-07-02:
@martinbudden thanks for the PRs.

I've been having a nightmare time the last couple of weeks, my schedule is clearing up now so I can get back to coding, finally!  Thanks for your patience, sorry things have been moving a little slowly.

I will merge the OSD code in, after which I will continue to work on the OSD code and I'm quite happy for you to continue working on and merging the betaflight mergebacks when they are ready as they are important for the 1.14 release.

martinbudden, 2016-07-02:
No worries. Good to hear that you've got everything sorted and are back to coding.

In the meantime I've been working on the new build system in betaflight - that's pretty well done now. Something to merge back into Cleanflight at some time in the future.

hydra, 2016-07-02:
yes i saw some comments regarding it but have not looked at it.  could you create an issue for that with a couple of links to some betaflight commits so we can track it.

hydra, 2016-08-10:
@martinbudden Any chance you can look at porting the more stable betaflight changes back in now that things have had a chance to settle?  Also should we create a ticket for the build system changes?

martinbudden, 2016-08-10:
@hydra , for quite a while now I've been focusing on some quite low-level stuff: changing the 'make' system in betaflight and then iNav and then converting iNav to the new IO. So I've lost sight a bit about what has been happening in the higher layers - PIDs and RC control. I think there have been some quite radical changes to both these things, but I'm not sure about the detail of those changes.

So I'm happy to help do this in the next release, but it will take be a bit of time to get back up to speed in those areas. One other thing is you need to decide what approach you want to take: take the betaflight changes wholesale (eg replace pidluxfloat with the betalfight PID), or just port over some of the changes.

There have also been quite a few changes to the ESC handling, ie oneshot,multishot etc, but again I haven't really followed those changes in detail.

There's also the F4 port, but that comes almost "for free" once all the new IO changes have been completed.

And yes, there should certainly be a ticket for the build system changes.

One thing I'd like to see is some reconvergence of the the betaflight, iNav and Cleanflight code bases. I see no reason, for example, why the `drivers`, `io` `rx` , `sensors` and `telemetry` directories should not be identical for all three forks (of course for this to happen Cleanflight needs to adopt the new IO and betaflight and iNav need to adopt parameter groups - but both those changes are on the roadmap). Indeed I can see a future where the changes are concentrated in the `fc` `flight` and `target` directories.

This convergence would reduce everyone's workload in the common areas, and so allow faster innovation by each fork in its area of speciality.

@borisbstyle , @digitalentity do you have any views on convergence?

hydra, 2016-08-23:
@martinbudden yes, we need convergence.  And I totally agree regarding the drivers/telemetry/rx/etc.  Perhaps the key to this is better modularisation to allow re-use of modules in other forks as required.

Dare I say it but perhaps repos for the modules are required?  I'm very wary of the overhead that might be introduced doing that but I feel the overhead would be LESS than the current overhead of continual merging and PR/code rot that we have in the 3 projects right now.

martinbudden, 2016-08-23:
I've also been thinking of a separate repo for the common parts. I think it is certainly feasible.

A prerequisite for this is that Cleanflight convert to the new IO. iNav has now done this and there is  (I estimate) >85% commonality between the drivers for iNav and betaflight.

There will probably always be some divergence, since betaflight experiments quite a bit in the driver area, but that can be accommodated.

hydra, 2016-08-24:
yup, agreed.  we need the new IO stuff ASAP.

hydra, 2016-08-24:
With regards to this issue, I will close it now. I want to get 1.14.0 out before some other changes are coming, i feel this issue is already out of date anyway.

closed 2016-08-24 by hydra

digitalentity, 2016-08-24:
@hydra please note that INAV is suffering from odd random glitches after IO has been merged. It's likely not related to the IO itself, but still, new IO might reveal long lurking bugs with stack and memory corruption - https://github.com/iNavFlight/inav/issues/431

martinbudden, 2016-08-24:
Yes, this issue is out of date. We can discuss betaflight mergebacks again once 1.14 is out.