cleanflight/cleanflight #2214: flight_pid_unittest failures on 32-bit

landstroem, 2016-06-06:
# Environment
$ (lsb_release -d && getconf LONG_BIT && make --version && gcc --version && g++ --version && make --version )
```
Description:	Ubuntu 14.04.4 LTS
32
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i686-pc-linux-gnu
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

g++ (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i686-pc-linux-gnu
```
Also tested on cygwin (32-bit w/ GCC 5.3.0) with the same results.

# Steps to reproduce
1. cd src/test
2. make test

# Expected results
All tests should pass like on the Travis CI build for the master branch.
https://travis-ci.org/cleanflight/cleanflight/branches

# Actual results
```
Running main() from gtest_main.cc
[==========] Running 9 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 9 tests from PIDUnittest
[ RUN      ] PIDUnittest.TestPidLuxFloat
unit/flight_pid_unittest.cc:277: Failure
Value of: unittest_pidLuxFloatCore_PTerm[FD_ROLL]
  Actual: 31.171875
Expected: calcLuxPTerm(pidProfile, FD_ROLL, rateErrorRoll)
Which is: 31.25
unit/flight_pid_unittest.cc:278: Failure
Value of: unittest_pidLuxFloatCore_ITerm[FD_ROLL]
  Actual: 0.36529541
Expected: ITermRoll
Which is: 0.36621094
unit/flight_pid_unittest.cc:280: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_ROLL]
  Actual: 15.5857
Expected: expectedDTerm
Which is: 15.624762
unit/flight_pid_unittest.cc:281: Failure
Value of: unittest_pidLuxFloatCore_PTerm[FD_PITCH]
  Actual: 31.171875
Expected: calcLuxPTerm(pidProfile, FD_PITCH, rateErrorPitch)
Which is: 31.25
unit/flight_pid_unittest.cc:282: Failure
Value of: unittest_pidLuxFloatCore_ITerm[FD_PITCH]
  Actual: 0.36529541
Expected: ITermPitch
Which is: 0.36621094
unit/flight_pid_unittest.cc:284: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_PITCH]
  Actual: 15.5857
Expected: expectedDTerm
Which is: 15.624762
unit/flight_pid_unittest.cc:285: Failure
Value of: unittest_pidLuxFloatCore_PTerm[FD_YAW]
  Actual: 66.240234
Expected: calcLuxPTerm(pidProfile, FD_YAW, rateErrorYaw)
Which is: 66.40625
unit/flight_pid_unittest.cc:286: Failure
Value of: unittest_pidLuxFloatCore_ITerm[FD_YAW]
  Actual: 0.54794312
Expected: ITermYaw
Which is: 0.54931641
unit/flight_pid_unittest.cc:288: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_YAW]
  Actual: 15.5857
Expected: expectedDTerm
Which is: 15.624762
unit/flight_pid_unittest.cc:295: Failure
Value of: unittest_pidLuxFloatCore_PTerm[FD_ROLL]
  Actual: 31.171875
Expected: calcLuxPTerm(pidProfile, FD_ROLL, rateErrorRoll)
Which is: 31.25
unit/flight_pid_unittest.cc:296: Failure
Value of: unittest_pidLuxFloatCore_ITerm[FD_ROLL]
  Actual: 0.73059082
Expected: ITermRoll
Which is: 0.73242188
unit/flight_pid_unittest.cc:298: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_ROLL]
  Actual: 15.5857
Expected: expectedDTerm
Which is: 15.624762
unit/flight_pid_unittest.cc:299: Failure
Value of: unittest_pidLuxFloatCore_PTerm[FD_PITCH]
  Actual: 31.171875
Expected: calcLuxPTerm(pidProfile, FD_PITCH, rateErrorPitch)
Which is: 31.25
unit/flight_pid_unittest.cc:300: Failure
Value of: unittest_pidLuxFloatCore_ITerm[FD_PITCH]
  Actual: 0.73059082
Expected: ITermPitch
Which is: 0.73242188
unit/flight_pid_unittest.cc:302: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_PITCH]
  Actual: 15.5857
Expected: expectedDTerm
Which is: 15.624762
unit/flight_pid_unittest.cc:309: Failure
Value of: unittest_pidLuxFloatCore_PTerm[FD_ROLL]
  Actual: 31.171875
Expected: calcLuxPTerm(pidProfile, FD_ROLL, rateErrorRoll)
Which is: 31.25
unit/flight_pid_unittest.cc:310: Failure
Value of: unittest_pidLuxFloatCore_ITerm[FD_ROLL]
  Actual: 1.0958862
Expected: ITermRoll
Which is: 1.0986328
unit/flight_pid_unittest.cc:312: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_ROLL]
  Actual: 15.5857
Expected: expectedDTerm
Which is: 15.624762
unit/flight_pid_unittest.cc:313: Failure
Value of: unittest_pidLuxFloatCore_PTerm[FD_PITCH]
  Actual: 31.171875
Expected: calcLuxPTerm(pidProfile, FD_PITCH, rateErrorPitch)
Which is: 31.25
unit/flight_pid_unittest.cc:314: Failure
Value of: unittest_pidLuxFloatCore_ITerm[FD_PITCH]
  Actual: 1.0958862
Expected: ITermPitch
Which is: 1.0986328
unit/flight_pid_unittest.cc:316: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_PITCH]
  Actual: 15.5857
Expected: expectedDTerm
Which is: 15.624762
unit/flight_pid_unittest.cc:323: Failure
Value of: unittest_pidLuxFloatCore_PTerm[FD_ROLL]
  Actual: 31.171875
Expected: calcLuxPTerm(pidProfile, FD_ROLL, rateErrorRoll)
Which is: 31.25
unit/flight_pid_unittest.cc:324: Failure
Value of: unittest_pidLuxFloatCore_ITerm[FD_ROLL]
  Actual: 1.4611816
Expected: ITermRoll
Which is: 1.4648438
unit/flight_pid_unittest.cc:326: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_ROLL]
  Actual: 15.5857
Expected: expectedDTerm
Which is: 15.624762
unit/flight_pid_unittest.cc:327: Failure
Value of: unittest_pidLuxFloatCore_PTerm[FD_PITCH]
  Actual: 31.171875
Expected: calcLuxPTerm(pidProfile, FD_PITCH, rateErrorPitch)
Which is: 31.25
unit/flight_pid_unittest.cc:328: Failure
Value of: unittest_pidLuxFloatCore_ITerm[FD_PITCH]
  Actual: 1.4611816
Expected: ITermPitch
Which is: 1.4648438
unit/flight_pid_unittest.cc:330: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_PITCH]
  Actual: 15.5857
Expected: expectedDTerm
Which is: 15.624762
[  FAILED  ] PIDUnittest.TestPidLuxFloat (20 ms)
[ RUN      ] PIDUnittest.TestPidLuxFloatIntegrationForLinearFunction
[       OK ] PIDUnittest.TestPidLuxFloatIntegrationForLinearFunction (1 ms)
[ RUN      ] PIDUnittest.TestPidLuxFloatIntegrationForQuadraticFunction
[       OK ] PIDUnittest.TestPidLuxFloatIntegrationForQuadraticFunction (0 ms)
[ RUN      ] PIDUnittest.TestPidLuxFloatITermConstrain
[       OK ] PIDUnittest.TestPidLuxFloatITermConstrain (1 ms)
[ RUN      ] PIDUnittest.TestPidLuxFloatDTermConstrain
unit/flight_pid_unittest.cc:489: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_ROLL]
  Actual: 15.5857
Expected: calcLuxDTerm(pidProfile, FD_ROLL, rateErrorRoll) / DTermAverageCount
Which is: 15.624762
unit/flight_pid_unittest.cc:507: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_ROLL]
  Actual: 1.5919757
Expected: calcLuxDTerm(pidProfile, FD_ROLL, rateErrorRoll) / DTermAverageCount
Which is: 1.5999757
unit/flight_pid_unittest.cc:517: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_ROLL]
  Actual: 9.5198545
Expected: calcLuxDTerm(pidProfile, FD_ROLL, rateErrorRoll) / DTermAverageCount
Which is: 9.5998535
[  FAILED  ] PIDUnittest.TestPidLuxFloatDTermConstrain (2 ms)
[ RUN      ] PIDUnittest.TestPidMultiWiiRewrite
[       OK ] PIDUnittest.TestPidMultiWiiRewrite (2 ms)
[ RUN      ] PIDUnittest.TestPidMultiWiiRewriteITermConstrain
[       OK ] PIDUnittest.TestPidMultiWiiRewriteITermConstrain (0 ms)
[ RUN      ] PIDUnittest.TestPidMultiWiiRewritePidLuxFloatCoreEquivalence
[       OK ] PIDUnittest.TestPidMultiWiiRewritePidLuxFloatCoreEquivalence (2 ms)
[ RUN      ] PIDUnittest.TestPidMultiWiiRewritePidLuxFloatEquivalence
unit/flight_pid_unittest.cc:768: Failure
Value of: unittest_pidLuxFloatCore_PTerm[FD_ROLL]
  Actual: 62.421875
Expected: calcLuxPTerm(pidProfile, FD_ROLL, rateErrorRollf)
Which is: 62.5
unit/flight_pid_unittest.cc:769: Failure
Value of: unittest_pidLuxFloatCore_ITerm[FD_ROLL]
  Actual: 0.73150635
Expected: ITermRoll
Which is: 0.73242188
unit/flight_pid_unittest.cc:771: Failure
Value of: unittest_pidLuxFloatCore_DTerm[FD_ROLL]
  Actual: 31.210463
Expected: expectedDTermf
Which is: 31.249523
[  FAILED  ] PIDUnittest.TestPidMultiWiiRewritePidLuxFloatEquivalence (1 ms)
[----------] 9 tests from PIDUnittest (38 ms total)

[----------] Global test environment tear-down
[==========] 9 tests from 1 test case ran. (39 ms total)
[  PASSED  ] 6 tests.
[  FAILED  ] 3 tests, listed below:
[  FAILED  ] PIDUnittest.TestPidLuxFloat
[  FAILED  ] PIDUnittest.TestPidLuxFloatDTermConstrain
[  FAILED  ] PIDUnittest.TestPidMultiWiiRewritePidLuxFloatEquivalence

 3 FAILED TESTS
```

hydra, 2016-06-11:
@martinbudden could you take a look, i confirm this here too.

martinbudden, 2016-06-12:
OK, I'll have a look. It's working fine on my system (Mac), so it's a bit odd. Also the figures seem to be out consistently by a factor of 1.0025 - so that number must arise out of something.

closed 2017-02-25 by hydra