cleanflight/cleanflight #2215: Motoer spinning up when calibrating escs via cleanflight

jeffwhu, 2016-06-07:
New build ZMR with Diatone v2 pdb,Naze 32 10 dof flashed with latest version of cleanflight, x4rsb receiver, micro minim osd (removed to use msp on usb) rotorgeeks rg20 escs.
Question......
When trying to calibrate escs via cleanflight without battery plugged in, I slide the master tab fully up and two motors start spinning with only the usb connected to pc.
https://youtu.be/jGPJBmQJJjM
could somebody help me to find out what happening.
Ceers
Jeff

Arakon, 2016-06-07:
If it's a naze rev. 6, that's the cause. Not related to cleanflight, but a hardware design issue.
https://hilo90mhz.com/naze-v6-reverse-voltage-feedback-issue-schottky-diode-fix/

jeffwhu, 2016-06-08:
Thanks Arakon, fitted the diode and working perfect now.

closed 2016-06-08 by jeffwhu