cleanflight/cleanflight #2217: Plethora fo system_stm32f30x .c and .h files

martinbudden, 2016-06-07:
Every F3 target includes the files `system_stm32f30x.c` and `system_stm32f30x.h` in its target directory (e.g. `/src/main/target/sparky/system_stm32f30x.c` etc). However they are all the same (except for the `NAZE32PRO` version.

There should only be a single copy of this file for all the targets, with an exception being made for the `NAZR32PRO`.

hydra, 2016-06-11:
likely we should just delete the naze32pro target too. since that target was never commercially released and is not even complete.

hydra, 2016-06-11:
Blocking by OSD support #2210 - that needs to happen first since there are makefile changes and it would complicate the merge.

martinbudden, 2016-07-08:
Fixed in PR https://github.com/cleanflight/cleanflight/pull/2288

closed 2016-07-08 by martinbudden