cleanflight/cleanflight #2220: Improvements to VBAT alarm settings

ethomas997, 2016-06-09:
With the current battery-warning settings it's difficult to configure precise voltage-alarm levels.  For instance, on a 3S setup, you can set "vbat_warning_cell_voltage = 35" for an alarm level of 10.5V, or "vbat_warning_cell_voltage = 34" for an alarm level of 10.2V, but nothing in between.

More resolution is needed on the 'vbat_max_cell_voltage', 'vbat_min_cell_voltage' and 'vbat_warning_cell_voltage' settings.  The best way I can think of to do this would be to support a new range of values, 1000 to 5000, corresponding to 1.000V to 5.000V on the settings.  The current range, 10 to 50, could still be allowed and supported for backward compatibility.  (A Configurator update would also be needed to support the new range on there.)

I can implement a PR to do this.  Thoughts?

--ET

hydra, 2016-06-11:
agreed, see #2221

hydra, 2016-06-11:
blocked by #2221

closed 2017-02-25 by hydra