cleanflight/cleanflight #2221: Cleanup battery code

hydra, 2016-06-10:
After working on the OSD code for a while it has come to my attention that the battery code, msp and configuration storage is far from ideal.

Issues:
1) There are dependency issues with throttleStatus_e. (only needed via virtual current meter)
2) All the configuration settings for current  meter, volt meter, battery capacity, cell voltages and alarm voltages are essentially mixed together.
3) alarm code is mixed in with battery code.

The FC has these requirements:

A volt meter (ADC input) which needs a scale for configuration.
A requirement to detect the amount of cells which needs min/max volt per cell for configuration.
A requirement to provide battery status (present, or not)
A requirement to beep when the battery is low or critical.
A physical current meter (ADC input) which needs some configuration.
A virtual current meter (throttle input) which needs some configuration.
A requirement to configure the FC via MSP.

Changes proposed:

Split voltage metering from battery code.
Split current meters so the different ones can be compiled independently.
Split cell detection and status code from voltage metering.
Update MSP commands for voltage/current metering and remove concept of a battery from them.
Add new MSP command to read and configure battery settings. (e.g. MIN/MAX volt per cell and capacity).

gaelj, 2016-06-11:
> Add new MSP command to read and configure battery settings. (e.g. MIN/MAX volt per cell and capacity).

It would be nice to be able to set the `mAhdrawn` value via MSP.

martinbudden, 2016-06-11:
Yes, the battery code is certainly confused at the moment. Splitting it as you describe would certainly be a good idea. The conglomeration of the virtual and the real current sensors has particularly grated with me.

ethomas997, 2016-06-12:
It'd be good if the VBATT_LPF_FREQ value was made configurable.  (I believe at this point Betaflight actually has a different hard-coded value than CF.)

hydra, 2016-09-11:
I pushed #2396 this evening which cleans up many of the issues.  More work still to do on the PR.  Review of PR welcomed.

hydra, 2016-09-12:
Done

closed 2016-09-12 by hydra