cleanflight/cleanflight #2223: GPS is unable to handle negative heights

fraternl, 2016-06-12:
Half of our country (the Netherlands) is beneath sea level. 
At home I'm at 20 metres above sea level, but at work in the province of Holland, Amsterdam, I'm flying 8 metres beneath sea level.

The negative value my uBlox reports to cleanflight is not handled properly and clips to zero, sometimes giving a large 16-bit integer.

stronnag, 2016-06-12:
sure it's not the case that CF handles it OK, and your visualisation application is misinterpreting as unsigned.

The MSP 'spec' misdefines it as unsigned in MSP_RAW_GPS, but this would only affect the  visualisation agent.

fraternl, 2016-06-12:
I think you're assuming I'm using an OSD. I'm not. I'm receiving the data though the sport interface to my Frsky.

Is the error made in that interface our is it wrong in OpenTX?

I have an openXsensor with GPS, but never took it to my work.

I could test that to check if the error is in sport.

stronnag, 2016-06-12:
The CF interface appears correct. It serialises a 16 integer, period. The signed / unsigned interpretation is up to the visualisation agent. I recall having to fix mwp (a GCS) for this because it stupidly assumed unsigned.

fraternl, 2016-06-13:
I just now tested it using the Chrome application Cleanflight configurator.
The height at the field near my work is reported as 65533 metres high...

It seems as if the error is made before it is given to the S-port or the error is made on more that one interface section.

fraternl, 2016-06-13:
./js/msp.js

- GPS_DATA.alt = data.getUint16(10, 1);
+ GPS_DATA.alt = data.getInt16(10, 1);

fraternl, 2016-06-13:
find -type f | xargs -I{} grep -i GPS_altitude {}
extern uint16_t GPS_altitude;              // altitude in 0.1m
uint16_t GPS_altitude;              // altitude in 0.1m
                            GPS_altitude = gps_Msg.altitude;
        GPS_altitude = _buffer.posllh.altitude_msl / 10 / 100;  //alt in m
            sbufWriteU16(dst, GPS_altitude);
            GPS_altitude = sbufReadU16(src);
    uint16_t altitude = GPS_altitude;
    ltm_alt = (sensors(SENSOR_SONAR) || sensors(SENSOR_BARO)) ? altitudeHoldGetEstimatedAltitude() : GPS_altitude * 100;
    ltm_alt = GPS_altitude * 100;
    uint16_t hottGpsAltitude = GPS_altitude + HOTT_GPS_ALTITUDE_OFFSET;   // GPS_altitude in m ; offset = 500 -> O m
                    smartPortSendPackage(id, GPS_altitude * 100); // given in 0.1m , requested in 10 = 1m (should be in mm, probably a bug in opentx, tested on 2.0.1.7)
        GPS_altitude * 1000,
        GPS_altitude * 1000,
        (sensors(SENSOR_SONAR) || sensors(SENSOR_BARO)) ? altitudeHoldGetEstimatedAltitude() * 10 : GPS_altitude * 1000,
        GPS_altitude * 1000,
        mavAltitude = GPS_altitude;
        mavAltitude = GPS_altitude;
#define HOTT_GPS_ALTITUDE_OFFSET 500
    {"GPS_altitude",      -1, UNSIGNED, PREDICT(0),          ENCODING(UNSIGNED_VB), CONDITION(ALWAYS)},
    blackboxWriteUnsignedVB(GPS_altitude);
    GPS_altitude = altitudeInMeters;
    EXPECT_EQ((int16_t)(hottGPSMessage->altitude_H << 8 | hottGPSMessage->altitude_L), 1 + HOTT_GPS_ALTITUDE_OFFSET);
uint16_t GPS_altitude;              // altitude in m
uint16_t GPS_altitude;              // altitude in 0.1m

stronnag, 2016-06-13:
So you have now found  a similar display bug in the configurator. Change
```` 
GPS_DATA.alt = data.getUint16(10, 1);
````
to 
````
GPS_DATA.alt = data.getInt16(10, 1);
````
and see what happens.

fraternl, 2016-06-13:
Yes..
But my problem is not the configurator.
It's the telemetry feedback through s.port

unsigned int is used there too

stronnag, 2016-06-13:
No your problem is the in the display, the encoding doesn't care.

fraternl, 2016-06-13:
Could you be more specific...   "the display" is used in a generic sense....

It can go wrong in a lot of places.  The S-port protocol itself, Cleanflight giving it to the S-Port, OpenTX interpreting the S-Port data.

In OpenTX I can have negative numbers for this telemetry item according to its settings.

schwabe, 2016-06-13:
This sounds like Cleanflight is transmitting a 16bit signed integer where it should transmit a 32 bit signed integer for the telemetry. The "normal" Barometer height works and both heights are handled in the same way in OpenTX as far as I can see.

schwabe, 2016-06-13:
int32_t BaroAlt = 0;
passing that into a uint32 will preserve the bitformat, while passing the 
int16_t (*1000) into a uint32 will not.

```
 #ifdef GPS
             case FSSP_DATAID_GPS_ALT    :
                 if (sensors(SENSOR_GPS) && STATE(GPS_FIX)) {
-                    smartPortSendPackage(id, GPS_altitude * 100); // given in 0.1m , requested in 10 = 1m (should be in mm, probably a bug in opentx, tested on 2.0.1.7)
+                    smartPortSendPackage(id, (int32_t) GPS_altitude * 100); // given in 0.1m , requested in 10 = 1m (should be in mm, probably a bug in opentx, tested on 2.0.1.7)
                     smartPortHasRequest = 0;
                 }
                 break;
```

should fix the problem

projectkk2glider, 2016-06-13:
While you are at it that comment can be corrected, the GPS altitude in SPORT format is in `cm` (centimeters).

ledvinap, 2016-06-13:
@schwabe : The problem is caused by GPS_altitude  being `uint16_t`. 
Before multiplication, it gets converted to int, mapping negative values to 65535-32768 interval.
It would be best to use `int16_t` (or int) for GPS_altitude.
using `(int16_t)GPS_altitude * 100` is possible temporrary workaround

fraternl, 2016-06-15:
Is anyone here able to actually implement the changes?

ledvinap, 2016-06-15:
@hydra: 

Current code related to GPS_altitude is mixing units, it is unclear  if it is in meters or decimeters (contradictory comments).
Meters is used int most of the code, some telemetry may not work correctly.

It would be best to change GPS_altitude to int32_t and match it's units with sonar and baro altitude.

Sonar and baro is in cm, personally I would prefer to use mm (mm will work to 2000km; neither cm or mm is enough for geostationary orbit, but I doubt something with CF onboard will get there soon ;-) )

What do you think?

martinbudden, 2016-06-15:
@hydra , @ledvinap , I am **strongly** in favour of suffixing **all** measurement variables with their units, eg `altitudeMm`, 'timeUs`, `timeMs`, `angleDeciDegrees` etc etc. This avoids simple errors and also allows what could be termed a form of dimensional analysis to spot more complex errors just by reading the code. But I seem to plough a lonely furrow on this issue...

closed 2017-02-25 by hydra

fraternl, 2017-02-27:
why is this ticket closed?
Is it fixed?

fraternl, 2017-02-28:
@hydra why did you close this ticket?

hydra, 2017-03-01:
all tickets were closed when the project was reset to use the BF 3.1 codebase. If after looking at the latest code you think this is still an issue then feel free to create a new issue that refers to this one.