cleanflight/cleanflight #2229: Blackbox for sensors only (without arming)

doisyg, 2016-06-13:
Hi,
I am interested in testing the accuracy of the different sensors of my board without attaching motors and ESCs. Is there an existing CLI command that would enable blackbox logging without arming?

Bonus question, to power the board without ESC from a 3.3V battery, I would need a DC-DC step up, right? There is no way of powering the board directly from 3.3V on any model?

martinbudden, 2016-06-15:
@doisyg , this is really a support question and should be asked in one of the rcgroups.

But to answer your question - why do you want to log without arming? You have no motors attached, so arming is harmless.

Closing.

closed 2016-06-15 by martinbudden

doisyg, 2016-06-15:
Okay, thanks, I will have a look at the rcgroups.
(I want to log without arming because I want to log without anything attached to the board but the battery)

ledvinap, 2016-06-15:
@doisyg : You may use RX_MSP and controller emulation in CF configurator to arm board ..
It is impossible to auto-arm the board (there is explicit code to disable auto-arming using channel)

ckwkalk, 2017-01-17:
@doisyg did you find a solution to this - I am interested in the sensor data also and would like to compare their "bench" performance without any motors etc.

@ledvinap does this mean that you would have to be connected to CleanFlight to log the data (in which case you could just use Tethered Logging)?

doisyg, 2017-02-06:
Going back to this. No I did not find any satisfying way of starting blackbox with nothing attached to the board.
Any suggestion on how to use a simple switch button? Like making a button acting as a receiver on one of the AUX input?