cleanflight/cleanflight #2236: Adopt improved target build support as implemented in betaflight

martinbudden, 2016-06-16:
@hydra , I don't know if you've noticed but I've been working with @blckmn , @AndersHoglund  and others to improve the way targets are handled in betaflight.

Now, for betaflight, all target changes happen the target directory. Source file changes (e.g. `pwmInit` and timer changes) go in `target.c` . `target.h` works as before. `Makefile` changes (e.g. list of source files, build flags, target type (F1, F3 etc)) goes in `target.mk`.

The system is working well and several target have been added to betaflight with no changes outside the `target` directory.

I've also implemented the system in 'iNav' and @digitalentity has added SPRacingF3 Evo support to iNav with no changes outside the `target` directory.

blckmn, 2016-06-16:
@hydra @martinbudden happy to assist in migrating these changes into CF if it is desired.

JohnOCFII, 2016-06-16:
I'd love to see FuryF3 support added, but hadn't mentioned it here since I recall that adding more targets was not high on the list due to extra testing efforts.  Perhaps this change would make that testing less difficult.

martinbudden, 2016-06-17:
@blckmn , you offer is appreciated. The 1.14 release is targeted for improved OSD support, but this is a strong candidate for the 1.15 release. BTW, I've enjoyed working with you on this.

@JohnOCFII , one of the advantages of this change is that it allows decentralised board support. If someone produces a new board they can do a Cleanflight fork, add the target directory build and test and  voila!, a new target. What's more, it's easy for them to do new releases to keep up to date with new Cleanflight releases.

In short: the burden of supporting a board is moved from Cleanflight to the board producer. Cleanflight developers have more time to enhance Cleanflight, and board developers aren't reliant on Cleanflight developers to merge in their code. Everybody wins. (Of course there will still be some cases where a board is so different that it requires a change to Cleanflight itself, but those cases should be minimised.)

blckmn, 2016-06-17:
@martinbudden it has been quite the pleasure - thanks.

1.15 sounds like a plan :)

I like the way it has panned out, especially as it promotes (as you say) a fork for a new target and removes the biggest impedance to that - the merging of updates.

closed 2017-02-25 by hydra