cleanflight/cleanflight #2238: Immersion Vortex 285 Altitude and GPS Hold

sag007, 2016-06-17:
Hi,

I have purchased the Vortex 285 and whilst the unit is good to fly, it would be awesome if it had altitude hold using the onboard barometer and an adon board for GPS hold.

I couldnt justify having two copters, one for photography and another for racing and would like to be able to use the Vortex for evreything because when the legs are folded in it very compact. 

Is any one working on adding these feature to the Vortex series of copters?

sag...

martinbudden, 2016-06-17:
You should look at iNav, the navigation-enabled fork of Cleanflight.

I'm not sure what flight controller is on the Vortex, so it might not be supported by iNav. However, support for more flight controllers is being added to iNav.

Eventually the navigation features in iNav will make it back into Cleanflight.

martinbudden, 2016-06-23:
Closing, since no further response from originator.

closed 2016-06-23 by martinbudden