cleanflight/cleanflight #2241: Baro Alt Hold problem

dacobi, 2016-06-19:
I'm using latest stable cleanflight on cc3d with MS5611.

I set 

alt_hold_deadband = 250
baro_cf_vel = 1 
baro_cf_alt = 0

Problem is, in alt hold mode, when I move throttle down and to absolute min my quad shoots upward.

SteveAmor, 2016-06-19:
Does it work as expected with an earlier version of code?

dacobi, 2016-06-19:
IIRC there were some generel instability issues with baro alt hold in the
code prior to 1.30.0?

But I haven't tried it with these exact setting. Will try tomorrow.

andre92, 2016-06-20:
@dacobi you are correct, CF < 1.13.0 had bugs in alt hold (and headfree) but it has been fixed in 1.13.0 stable. (was bugged from 1.12.0 "stable" onwards, reverting will not help you)
may I see a full CLI dump?
I suppose you left alt_hold_fast_change = ON (as default)
Those 3 settings you changed work fine as stock:
set baro_cf_vel =  0.985
set baro_cf_alt =  0.965
set alt_hold_deadband = 50

If you have a noisy frame you could try lowering acc_cut_hz below 15 (lower is less noise, do not raise it above 15, it is opposite to the old acc_lpf_factor)

I think you are using a too large deadband combined with alt_hold_fast_change.

250 for alt_hold_deadband is way too large, especially if alt_hold_fast_change is ON.
inside the deadband, it tries to hold altitude.
outside the deadband it will either control velocity (alt_hold_fast_change = OFF), or set the throttle directly (alt_hold_fast_change = ON, which is default)
with a deadband this large, you can only use top and bottom end of throttle.

dacobi, 2016-06-20:
I did leave alt_hold_fast_change = ON, but I'm still pretty sure that even
with throttle at absolute minimum the quad still shoots up. (I can test in
a couple of hours)

One thing I'm having trouble understanding is: baro_cf_alt = 0.965 (default)

If 1.0 is accel only? Why such a high default value?

andre92, 2016-06-20:
@dacobi correct, 1.0 means accel only, 0.0 means baro only. baro gives noise, accel gives drift but without noise. 
The default is probably set so high to avoid any and all baro noise with some acceptable accel drift.
You can lower it until you see noise, then slightly raise. 
I didn't mess with it because the default works for me, after I set acc_cut_hz to 10. (noisy frame)

by the way, what kind of quad do you have, and how much throttle do you need to give for hover? My 450 hovers around center.

dacobi, 2016-06-20:
I'm not sure I know what kind of noise you mean. Mechanical? Electrical?

My quad is a 280 racing, I think it hovers at around 40%

andre92, 2016-06-20:
In the case of an accelerometer, most noise is mechanical, eg. vibrations from motors and props.
This can be reduced by soft mounting the flight controller or lowering acc_cut_hz.
I have mine hard mounted and lowered acc_cut_hz to 10 which gives me acceptable noise levels.
(for a racer which has higher frequency vibrations, the default 15 should be good)
In the case of a barometer, noise can come from wind, turbulence or light falling into the sensor. You can see the amount of noise on the graph.
Since you covered your barometer, it should be noise free. You can check it in the sensor graph. Smooth sensor curves with no notable spikes means it is noise free.

closed 2017-02-25 by hydra