cleanflight/cleanflight #2244: Sparky 1 with MPU-9250 / MPU-6500 via I2C

x0054, 2016-06-21:
First, I would like to apologize as this problem is entirely self created :( I used Sparky 1 Schematics to make a board for a tiny drone I am building. The schematics are mostly identical except I replace MPU-9150 with MPU-9250, though the MPU-9250 is still connected via I2C. The rest is the same. After successfully flashing in DFU mode I can't connect and get 2 long LED flashes before rapid 4 flashes (reset?), from what I understand an ACC error.

MPU-9250 uses MPU-6500 instead of MPU-6050 for ACC, however, I am still using I2C for connection. I have been digging around the code for the last 5 hours, but I can't figure out what I need to change in target.h file. Could someone point me in the right direction. I can compile the code and test no problem. Again, sorry, I know this is self created problem. Thank you.

x0054, 2016-06-21:
I was able to connect to the board after changing the target settings to fake ACC and GYRO. However, defining MPU6500 in the target.h like so:

`#define ACC
#define USE_ACC_MPU6500
#define ACC_MPU6500_ALIGN CW270_DEG`

throws an error during compile:

`/src/main/sensors/initialisation.c:451: undefined reference to `mpu6500AccDetect'`

mpu6500AccDetect is defined in `drivers/accgyro_mpu6500.h` and that header is included in initialisation.c, so I can't figure out what's going on :(

ledvinap, 2016-06-21:
accgyro_mpu6500.o is probably not linked in for Sparky, you must add it in makefile

x0054, 2016-06-21:
Thank you!!!! Why on earth did I forget to do that!? I feel like such an idiot!

closed 2016-06-21 by x0054