cleanflight/cleanflight #2247: CC3D settings not being saved/crashing board.

Z-slim, 2016-06-21:
I am having issues with CF and my CC3D. Everything was fine, and flashed to latest version (no OP BL) no problem. Connects and works with CF. This is where it gets weird and unstable...
I can change some settings via GUI or CLI and they save between reboots without issue. These would be TELEMETRY, RX_SERIAL (SBUS), SONAR, BLACKBOX & MOTOR_STOP. This indicates that the connection is in fact ok and settings can be manipulated successfully as seen in the receiver tab with all tx sliders moving properly. When I try to disable feature ONESHOT125, or modify the looptime, both settings will not save (yes, I am saving...) between boots and are back to their original values upon next boot and shows as much in both GUI and CLI (loop=1000, oneshot125=on). Even stranger is when I enable SERVO_TILT, the USB VCP crashes and I can no longer connect to CF as it shows up as an unrecognized device to windows. I have to fire up my FTDI and reflash via CC3D Mainport to regain access... as you can imagine... all settings default and I have to do setup all over again!!! 

Am I doing something wrong or just better off sticking with LibrePilot? I really don't want this to ruin my experience with CF as I've heard so many great things about it, just not seeing them...   This all happening on a CC3D with no other hardware connected other than USB & SBUS. All of this worked great in LP, with the exception of sonar (not supported).

Thanks for any ideas or solutions, my eyes and brain are tired from all the searching and reading, but can't find anything that helps.

Z-slim, 2016-06-27:
Reflashing via configurator download/installed firmware (1.13) resolved the issues and board is 100% functional. Had manually retrieved 1.13 from repository before, may have been corrupt somehow during D/L for the odd behavior. Other flashes have been fine before this way, oh well... Loving CF so much more than LibrePilot (though, be fair, its great in its own right for OP boards).

closed 2016-06-27 by Z-slim

dohdohpub, 2016-08-30:
Hello!
I got the same problem with Betaflight 3.0 when I set up SERVO_TiILT.
https://www.youtube.com/watch?v=KrikfRPmjiA
What is the conflict of the system setting SERVO_TILT?